Ardulink/Ardulink-2 #1: ardulink-core-mqtt prevents resolving dependencies

fantasticfears, 2016-04-27:
I am looking forward to compare several serial communication protocol for Arduino. Ardulink2 looks promising to me. 

But I found `ardulink-core-mqtt` is preventing the building.

I use Intellij IDEA 2016.1.1. Removing several related packages with it can make it possible to run. But I suppose there is a problem which I don't understand.

I attempted to put a bintray https url, took a partial snippet from bintray website and updating to the latest 0.7 for moquette-broker. Though none of those can work :(

https://bintray.com/andsel/maven/org.dna.mqtt.moquette-broker/view

pfichtner, 2016-04-27:
What type of error do you get? Can you paste a snippet somewhere? 
A quickfix could be to purge the ardulink-core-mqtt from the parent pom.

fantasticfears, 2016-04-28:
I clear builds by `clean`. And another dependency issue comes up..

I ran the `compile` on parent:

```
/Library/Java/JavaVirtualMachines/jdk1.8.0_60.jdk/Contents/Home/bin/java "-Dmaven.home=/opt/homebrew-cask/Caskroom/intellij-idea/2016.1.1/IntelliJ IDEA.app/Contents/plugins/maven/lib/maven3" "-Dclassworlds.conf=/opt/homebrew-cask/Caskroom/intellij-idea/2016.1.1/IntelliJ IDEA.app/Contents/plugins/maven/lib/maven3/bin/m2.conf" -Didea.launcher.port=7534 "-Didea.launcher.bin.path=/opt/homebrew-cask/Caskroom/intellij-idea/2016.1.1/IntelliJ IDEA.app/Contents/bin" -Dfile.encoding=UTF-8 -classpath "/opt/homebrew-cask/Caskroom/intellij-idea/2016.1.1/IntelliJ IDEA.app/Contents/plugins/maven/lib/maven3/boot/plexus-classworlds-2.4.jar:/opt/homebrew-cask/Caskroom/intellij-idea/2016.1.1/IntelliJ IDEA.app/Contents/lib/idea_rt.jar" com.intellij.rt.execution.application.AppMain org.codehaus.classworlds.Launcher -Didea.version=2016.1.1 --update-snapshots compile
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] parent
[INFO] ardulink-core-util
[INFO] ardulink-core-beans
[INFO] ardulink-core-base
[INFO] ardulink-core-serial-rxtx
[INFO] ardulink-core-bluetooth
[INFO] ardulink-core-digispark
[INFO] ardulink-core-raspberry
[INFO] ardulink-core-proxy
[INFO] ardulink-core-mqtt
[INFO] ardulink-legacy
[INFO] ardulink-networkproxyserver
[INFO] ardulink-mqtt
[INFO] ardulink-swing
[INFO] ardulink-console
[INFO] ardulink-mail
[INFO] example-datareceiver
[INFO] example-joysticksmartcardriver
[INFO] example-simplesmartcardriver
[INFO] deploy-dist
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building parent 2.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building ardulink-core-util 2.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-resources-plugin:2.5:resources (default-resources) @ ardulink-core-util ---
[debug] execute contextualize
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /Users/fantasticfears/Dev/Ardulink-2/ardulink-core-util/src/main/resources
[INFO] 
[INFO] --- maven-compiler-plugin:2.3.2:compile (default-compile) @ ardulink-core-util ---
[INFO] Nothing to compile - all classes are up to date
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building ardulink-core-beans 2.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-resources-plugin:2.5:resources (default-resources) @ ardulink-core-beans ---
[debug] execute contextualize
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /Users/fantasticfears/Dev/Ardulink-2/ardulink-core-beans/src/main/resources
[INFO] 
[INFO] --- maven-compiler-plugin:2.3.2:compile (default-compile) @ ardulink-core-beans ---
[INFO] Nothing to compile - all classes are up to date
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building ardulink-core-base 2.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-resources-plugin:2.5:resources (default-resources) @ ardulink-core-base ---
[debug] execute contextualize
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 1 resource
[INFO] 
[INFO] --- maven-compiler-plugin:2.3.2:compile (default-compile) @ ardulink-core-base ---
[INFO] Nothing to compile - all classes are up to date
[... 157 lines cut ...]
	at org.sonatype.aether.impl.internal.DefaultArtifactResolver.resolveArtifacts(DefaultArtifactResolver.java:216)
	at org.sonatype.aether.impl.internal.DefaultArtifactResolver.resolveArtifact(DefaultArtifactResolver.java:193)
	at org.apache.maven.repository.internal.DefaultModelResolver.resolveModel(DefaultModelResolver.java:122)
	... 39 more
Caused by: org.sonatype.aether.transfer.ArtifactNotFoundException: Failure to find org.dna.mqtt:moquette-parent:pom:0.6 in http://dl.bintray.com/andsel/maven/ was cached in the local repository, resolution will not be reattempted until the update interval of bintray has elapsed or updates are forced
	at org.sonatype.aether.impl.internal.DefaultUpdateCheckManager.newException(DefaultUpdateCheckManager.java:230)
	at org.sonatype.aether.impl.internal.DefaultUpdateCheckManager.checkArtifact(DefaultUpdateCheckManager.java:204)
	at org.sonatype.aether.impl.internal.DefaultArtifactResolver.resolve(DefaultArtifactResolver.java:427)
	... 42 more
[ERROR] 
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

Process finished with exit code 1
```

----

I guess the purge is `dependency:purge-local-repository`. It did success on parent but takes no effect.

Related Maven settings:

[image]

pfichtner, 2016-04-30:
On a fresh checkout I got the same errors. @LucianoZu could you please fix them?

LucianoZu, 2016-04-30:
Ok I'll see it asap

LucianoZu, 2016-05-02:
Now it should be fixed. The problem I encountered was on MQTT for a JAR used in tests (moquette). Bluetooth compiles like a charms. Maybe some push solved the issue on bluetooth.

Now Travis CI is up and running.

LucianoZu, 2016-05-02:
If you agree please close this issue

fantasticfears, 2016-05-03:
It works perfectly on my end. Thanks!

closed 2016-05-03 by fantasticfears