Ardulink/Ardulink-2 #4: Migrate mqtt to apache camel

pfichtner, 2016-05-05:
Instead of directly referring  mqtt third party mqtt libraries we could integrate into apache camel like we did with ardulink-mail.

LucianoZu, 2016-05-12:
Yes, we should define a module software with the generic Camel Endpoint for Ardulink (taking it from ardulink-mail) and then use it with ardulink-mail and ardulink-mqtt

pfichtner, 2017-08-14:
I started working on that last week

pfichtner, 2017-08-26:
Mqtt now using camel but still some work to do because "Ardulink camel" component was not generalized but copied. So some unification work has to be done.

pfichtner, 2017-08-28:
All done! Now there is a generic Ardulink-Camel component which is used by Mqtt and Mail. A Rest integration would now be easy to implement.

closed 2017-08-28 by pfichtner

pfichtner, 2023-08-06:
Supports #5 and can be closes with next release