Ardulink/Ardulink-2 #7: connection notifies

LucianoZu, 2016-05-07:
connection behavior has to be reviewed. Actually Console connect button doesn't work as expected

pfichtner, 2016-05-08:
confirmed, will fix it

LucianoZu, 2016-05-08:
I think could be useful that Link interface has the method isClosed (or isConnected) to help status management on setLink methods of Linkable interface.
Actually new Link interface hasn't connect method as the old one but maybe is not required to have.

pfichtner, 2016-05-08:
A created Link is an open connection created of the attributes of a config object. If it's broken it's eay to create a new one. Certainly not all link types will be able to handle connection state, e.g. notice if the underlying connection is broken. So I didn't want to mess the general Link interface with things not all Link types are able to fulfill. I'll take a look what could be a possible solution.

pfichtner, 2016-05-10:
@LucianoZu could you please verify if the issue was fixed? (should work since yesterday)

LucianoZu, 2016-05-12:
Ok, it works very fine. I've changed just a little enabling/disabling components code

closed 2016-05-12 by LucianoZu