Ardulink/Ardulink-2 #10: Console serial link with pingprobe doesn't work

LucianoZu, 2016-05-12:
From the console with serial link if pingprobe property is selected an exception is thrown and console doesn't permit connections anymore.

java.lang.IllegalStateException: Waited for arduino to boot but no response received
	at org.ardulink.util.Preconditions.checkState(Preconditions.java:55)
	at org.ardulink.core.serial.rxtx.SerialLinkFactory.waitForArdulink(SerialLinkFactory.java:86)
	at org.ardulink.core.serial.rxtx.SerialLinkFactory.newLink(SerialLinkFactory.java:73)
	at org.ardulink.core.serial.rxtx.SerialLinkFactory.newLink(SerialLinkFactory.java:1)
	at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:466)
	at org.ardulink.gui.ConnectionPanel.createLink(ConnectionPanel.java:231)
	at org.ardulink.gui.Console.createLink(Console.java:311)
	at org.ardulink.gui.Console$5.actionPerformed(Console.java:301)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6535)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

LucianoZu, 2016-05-12:
Maybe sketch has to send a message just after Serial initialization.

pfichtner, 2016-05-14:
Cannot reproduce. Works for me when using ArdulinkProtocol.ino

LucianoZu, 2016-05-15:
I added probe message in other sketches. Unfortunately for Uno32 still not work.

LucianoZu, 2016-10-20:
Maybe this bug is linked to #14

LucianoZu, 2017-01-26:
After timeout message from sketch arrives but it's too late. It seems that stream lock until Ardulink waits for it!
Next connection doesn't work anymore (port busy with both RXTX and JSSC)

LucianoZu, 2017-01-29:
Peter please test pingprobe with Linux to confirm the resolution bug

closed 2017-01-29 by LucianoZu