Ardulink/Ardulink-2 #12: Move Generic Connection Panel

LucianoZu, 2016-05-12:
Generic Connection Panel should be a generic component in ardulink-swing.
Examples already use it.

@pfichtner if you agree I'll do it asap.

pfichtner, 2016-05-13:
Do you mean ConnectionPanel and GenericPanelBuilder? Or just ConnectionPanel? Anyway: If one or both of them can be reused by other components feel free to move them to ardulink-swing.

LucianoZu, 2016-05-13:
Ok, I mean all the classes used for build the connection screen on console. Well I'll do it asap

LucianoZu, 2016-05-13:
I had to copy some classes for test unit (for example DummyLinkFactory or BaseBuilder). Peter if this is not the best practice, please leave to me a comment or move this issue to done status.

pfichtner, 2016-05-14:
Classes in test are things that test classes/units in the same maven module. If the system under test (SUT) gets moved those tests have to been moved, too. There is absolutely no need to duplicate things. I deleted the corresponding classes.

closed 2016-05-14 by pfichtner