Ardulink/Ardulink-2 #13: Exclude ardulink-core-virtual from assembly

pfichtner, 2016-05-14:
ardulink-core-virtual should not get distributed so it should be excluded in assembly.xml

LucianoZu, 2016-05-14:
Why? If we exclude it from assembly then we have to remove it from ardulink-console pom.xml

pfichtner, 2016-05-14:
-Who did add it to ardulink-console pom.xml?
-Why was it added there?
-If so: For the moment the poms do not have any impact on the dist zip, do they?
-All the runtime deps in ardulink-console pom.xml should be marked as optional
-Why shall we distribute ardulink-core-virtual? It was intended for testing purposes

pfichtner, 2016-05-14:
I see, it was added by you. My intention was to not add it.

pfichtner, 2016-05-14:
I did mark the runtime deps of ardulink-console as optional

LucianoZu, 2016-05-14:
Yes, I added to try it with swing components. It work very well and I
supposed that is not a great problem add it to distribution. We can remove
it from pom (or mark it as optional) without problem.

We are a two members... so if a thing is not done from you... 😉

pfichtner, 2016-05-14:
If we don't exclude it in assembly.xml it will be distributed. Do you want to distribute it? Distribution means it will be available in console's drop down combo. In my opinion this does not make any sense.

LucianoZu, 2016-05-14:
No, I agree with you.