Ardulink/Ardulink-2 #14: Disconnecting the serial link hangs

pfichtner, 2016-05-14:


LucianoZu, 2016-05-14:
Do you mean unplugging the usb cable?

pfichtner, 2016-05-14:
No, pushing the disconnect button

LucianoZu, 2016-05-15:
On windows systems it works very well. I tested it with a chipkit uno32 (but I think it works also with other boards). Of course when you press connect for the second time the board reboots.

So:
1. connect
2. send some messages
3. disconnect
4. connect
5. board reboots
6. send some other messages

it works.

while if after connection I unplug the usb cable I have these error messages (one each second)

Error 0x5 at ..\rxtx\src\termios.c(2712): Access denied.

and connection doesn't work anymore.

I tried also with a digispark (with another connection type) it works even better because the board doesn't reboot.

pfichtner, 2016-05-16:
This is not an Ardulink-2 issue but a general rxtxt issue. Also with older versions of Ardulink (back til 0.4.1) I got the same problem. 
@LucianoZu what to do with the issue? 

For reference, RXTXPort#close does not complete because IOLocked is non-zero
```
while( IOLocked > 0 )
{
	if( debug )
		z.reportln("IO is locked " + IOLocked);
	try {
		Thread.sleep(500);
	} catch( Exception e ) {}
}

```

pfichtner, 2016-05-16:
I did add https://github.com/Ardulink/Ardulink-2/tree/master/ardulink-core-serial-jssc which seems to work (could switch digital pin) and does not have the disconnect issue on my machine. 
Another advantage of jscc is that it manages the native libraries internally. 

I did add ardulink-core-serial-jssc as an alternative serial module so at the moment to serial modules are available (serial and serial-jssc). We should prefer one of them and purge the unneeded at least from console's runtime deps.

LucianoZu, 2016-05-16:
Yes, It was an issue in Ardulink1 too.
Good so we have another serial implementation to use. In Console maybe we
can use both since it is also a test tool for users.

LucianoZu, 2017-01-29:
This is an RXTX issue not JSSC that works very fine (on windows). So JSSC for its simpler use and its more reliable behavior is the preferred serial connection method.

pfichtner, 2024-11-29:
RXTX seems to be unmaintained, will remove RXTX support from Ardulink in the future. 
Just added [jserialcomm](https://fazecast.github.io/jSerialComm/) which could get released with the next version. 
In the future we could migrate to [purejavacomm](https://github.com/nyholku/purejavacomm). it does not depend on native libs but would require at least Java 11

closed 2024-11-29 by pfichtner