Ardulink/Ardulink-2 #15: Ardulink-2 examples

themarpe, 2016-10-01:
Hello
I tried to find some example code to connect to an arduino device with jssc, but I can't seem to find any.
I did compile the source so I think the libraries should work but I'm kind of lost around the documentation of any sort for the ardulink-2.
Would anyone be kind enough to share an example or two with me for how to connect to a certain COM port with the jssc and ardulink-2.
Thank you.

Best Regards, Martin

LucianoZu, 2016-10-04:
Hi Martin,
we are working on a Ardulink 2 stable version. Actually we are very near and we are going to merge camel_openQCM_peter branch to master.

So in the meanwhile I suggest you to try with camel_openQCM_peter branch.

There are a couple of examples that should work. Maybe you are searching for this:
org.ardulink.DataReceiver.createLink() that you can find here:
https://github.com/Ardulink/Ardulink-2/tree/master/example-datareceiver/src/main/java/org/ardulink

themarpe, 2016-10-06:
Thank you, I did get list of ports and was able to connect to a certain port using jssc library.
I will wait for the stable release. Do you plan on releasing any documentation along with it?

LucianoZu, 2016-11-05:
Ardulink 2 is now available to download. There are into the distro several examples. In our site you can find a couple of articles to get started with.

Soon (maybe in few hours) we will release a patch release with some bug fixing and another example.

closed 2016-11-05 by LucianoZu