Ardulink/Ardulink-2 #17: ardulink-mqtt memory leak?

mihalski, 2016-11-30:
It seems I may have run into a bug. I was running ardulink-mqtt on a Windows 10 machine with an arduino with a PIR sensor attached and  I noticed that nothing had been logged since last night. I hit ctrl-c at the ardulink-mqtt command line and rather than quitting it suddenly blasted through hundreds of entries as if they had buffered up, This was also visible at the MQTT broker.

Here’s a sample of what this looked like:

```
Nov 29, 2016 11:32:59 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get false
Nov 29, 2016 11:33:03 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get true
Nov 29, 2016 11:33:07 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get false
Nov 29, 2016 11:33:10 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get true
Nov 30, 2016 9:34:38 AM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get false
Nov 30, 2016 9:34:38 AM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get true
Nov 30, 2016 9:34:38 AM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get false
Nov 30, 2016 9:34:38 AM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D2/value/get true
```

I then ran into the following issue when trying to run it a bit later.. It is almost immediate:
```
C:\Users\michal\Downloads\ardulink-V2.0.1-20161105-1901\ardulink\lib>java -jar ardulink-mqtt-2.0.1.jar -brokerHost homepi -d 3 -connection ardulink://serial?port=COM10
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/Users/michal/Downloads/ardulink-V2.0.1-20161105-1901/ardulink/lib/slf4j-jdk14-1.7.12.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Users/michal/Downloads/ardulink-V2.0.1-20161105-1901/ardulink/lib/slf4j-log4j12-1.7.5.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
Nov 30, 2016 7:21:31 PM org.ardulink.mqtt.MqttMain$MqttClient connect
INFO: Connected to mqtt broker
Nov 30, 2016 7:21:31 PM org.ardulink.core.ConnectionBasedLink startListening
INFO: Starting listening on pin DIGITAL 3
Nov 30, 2016 7:21:31 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D3/value/get true
Nov 30, 2016 7:21:33 PM org.ardulink.mqtt.MqttMain$MqttClient fromArduino
INFO: Publishing arduino state change home/devices/ardulink/D3/value/get false
Exception in thread "hawtdispatch-DEFAULT-1" java.lang.OutOfMemoryError: Java heap space
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:35)
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:25)
        at org.fusesource.hawtdispatch.util.ThreadLocalPool.checkout(ThreadLocalPool.java:78)
        at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.read(AbstractProtocolCodec.java:310)
        at org.fusesource.hawtdispatch.transport.TcpTransport.drainInbound(TcpTransport.java:706)
        at org.fusesource.hawtdispatch.transport.TcpTransport$9.run(TcpTransport.java:770)
        at org.fusesource.hawtdispatch.internal.SerialDispatchQueue.run(SerialDispatchQueue.java:100)
        at org.fusesource.hawtdispatch.internal.pool.SimpleThread.run(SimpleThread.java:77)
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:35)
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:25)
        at org.fusesource.hawtdispatch.util.ThreadLocalPool.checkout(ThreadLocalPool.java:78)
        at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.read(AbstractProtocolCodec.java:310)
        at org.fusesource.hawtdispatch.transport.TcpTransport.drainInbound(TcpTransport.java:706)
        at org.fusesource.hawtdispatch.transport.TcpTransport$6.run(TcpTransport.java:588)
        at org.fusesource.hawtdispatch.internal.NioDispatchSource$3.run(NioDispatchSource.java:209)
        at org.fusesource.hawtdispatch.internal.SerialDispatchQueue.run(SerialDispatchQueue.java:100)
        at org.fusesource.hawtdispatch.internal.pool.SimpleThread.run(SimpleThread.java:77)
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:35)
        at org.fusesource.hawtdispatch.util.BufferPool.create(BufferPool.java:25)
        at org.fusesource.hawtdispatch.util.ThreadLocalPool.checkout(ThreadLocalPool.java:78)
        at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.allocateNextWriteBuffer(AbstractProtocolCodec.java:153)
        at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.flushNextWriteBuffer(AbstractProtocolCodec.java:186)
        at org.fusesource.hawtdispatch.transport.AbstractProtocolCodec.flush(AbstractProtocolCodec.java:229)
        at org.fusesource.hawtdispatch.transport.TcpTransport.flush(TcpTransport.java:674)
[... 15 lines cut ...]
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
```
And it just continues like that for MANY entries before occasionally getting a few more mqtt updates in. 

Regards,
Michal

mihalski, 2016-12-01:
In addition I've since had this:

```
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2" java.lang.OutOfMemoryError: Java heap space
Exception in thread "hawtdispatch-DEFAULT-2"
Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "hawtdispatch-DEFAULT-2"
Exception in thread "hawtdispatch-DEFAULT-2" Exception in thread "hawtdispatch-DEFAULT-1" java.lang.OutOfMemoryError: Java heap space
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x5f43631e, pid=7768, tid=0x00004674
#
# JRE version: Java(TM) SE Runtime Environment (8.0_111-b14) (build 1.8.0_111-b14)
# Java VM: Java HotSpot(TM) Client VM (25.111-b14 mixed mode windows-x86 )
# Problematic frame:
# V  [jvm.dll+0xe631e]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\michal\Downloads\ardulink-V2.0.1-20161105-1901\ardulink\lib\hs_err_pid7768.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

C:\Users\michal\Downloads\ardulink-V2.0.1-20161105-1901\ardulink\lib>
```

[hs_err_pid7768.log.txt](https://github.com/Ardulink/Ardulink-2/files/625705/hs_err_pid7768.log.txt)

I'm also running this on Mac OS 10.11 for a few days straight without a problem, so it seems to be Windows specific.

Regards,
Michal

LucianoZu, 2016-12-02:
I can't reproduce the bug with Windows 7 or Windows 10. It seems a mqtt-client bug from https://github.com/fusesource/mqtt-client

I've done a dependency update to the last mqtt-client version available. Mihalski can you try this version downloading it from here?
http://arduinopclink.altervista.org/javadoc/ardulink-V2.0.2-SNAPSHOT-20161202-2125.zip

pfichtner, 2016-12-03:
I am just monitoring memory usage of the current version with two analog and two digital listeners using the "virtual" connection with MqttMain's standalone feature where also the broker runs inside the same VM. Will give it some hours.

mihalski, 2016-12-04:
Not sure what's going on here:
```
C:\Users\michal\Downloads\ardulink\lib>java -jar ardulink-mqtt-2.0.2-SNAPSHOT.jar -brokerHost homepi -connection ardulink://serial?port=COM10 -d 2
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/Users/michal/Downloads/ardulink/lib/slf4j-jdk14-1.7.12.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Users/michal/Downloads/ardulink/lib/slf4j-log4j12-1.7.5.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
Exception in thread "main" java.lang.NoClassDefFoundError: org/fusesource/mqtt/client/MQTT
        at org.ardulink.mqtt.MqttMain$MqttClient.newClient(MqttMain.java:152)
        at org.ardulink.mqtt.MqttMain$MqttClient.<init>(MqttMain.java:119)
        at org.ardulink.mqtt.MqttMain$MqttClient.<init>(MqttMain.java:108)
        at org.ardulink.mqtt.MqttMain.connectToMqttBroker(MqttMain.java:296)
        at org.ardulink.mqtt.MqttMain.doMain(MqttMain.java:278)
        at org.ardulink.mqtt.MqttMain.main(MqttMain.java:265)
Caused by: java.lang.ClassNotFoundException: org.fusesource.mqtt.client.MQTT
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 6 more
```

LucianoZu, 2016-12-04:
My fault! I've specified mqtt-client 1.14 in Ardulink MQTT but not in Ardulink core MQTT (our mqtt client) so build was not coherent.

please retry here:
[](http://arduinopclink.altervista.org/javadoc/ardulink-V2.0.2-SNAPSHOT-20161204-1448.zip)

LucianoZu, 2016-12-04:
http://arduinopclink.altervista.org/javadoc/ardulink-V2.0.2-SNAPSHOT-20161204-1448.zip

pfichtner, 2016-12-04:
Can't reproduce the problem using official ardulink-2 release running on Linux. Ardulink-2 did run fine for more than 24 hours now monitored using visualvm.

pfichtner, 2016-12-06:
Played around a little bit more: Ardulink did run now for more than 24 hours publishing events to a separate broker. I did stop the broker for several hours to verify this won't cause the problem. But even after restarting the broker I could not reproduce a memory leak.

mihalski, 2016-12-07:
I'll get a chance to test the updates version on windows today, but it is worth noting that on OS X I either did not have a memory leak, or simply did not experience the same symptoms as that's been running perfectly since I first started it. It looks to be a windows specific thing even though it's running in a java virtual machine.

LucianoZu, 2017-01-19:
Upgraded to mqtt-client 1.14 on master branch. No other feedback available. Closing this issue.

closed 2017-01-19 by LucianoZu