Ardulink/Ardulink-2 #18: Waited for arduino to boot but no response received

GandalfD, 2016-12-29:
I set up the link using:
`Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=9600&baudrate=9600"));`
and it seems to connect to the arduino successfully. After about 5 seconds however, i get this error:
`Exception in thread "main" java.lang.IllegalStateException: Waited for arduino to boot but no response received`

When I run the program, the TX lights on my arduino blink for 5 seconds or so, which means that it finds the connection successfully. Do I need to make a sketch that replies to the computer once my program runs?

Thanks

GandalfD, 2016-12-29:
Whoops, turns out that I just needed to add "&pingprobe=false" in the link URI. That fixed everything

closed 2016-12-29 by GandalfD

LucianoZu, 2016-12-29:
Yes,
it depends on how you have configured uploaded sketch. In other words if you have something like this:

  Serial.print("alp://rply/");
  Serial.print("ok?id=0");
  Serial.print('\n'); // End of Message
  Serial.flush();

after your Serial.begin(9600) then you could have pingprobe=true (that is the default value). I've noted that in windows systems this could not work so with pingprobe=false the problem is definitely solved.

I've written baudrate 9600 accordingly with your connection URI (default value is 115200). However your port is very strange. It should appear as COM<number> in Windows systems or /dev/tty<something> in Linux systems.