Ardulink/Ardulink-2 #19: Port Busy

elpmoraes, 2017-01-09:
Hello !!

I'm starting to study ardulink on intellij, but i always receive this exception:

> Exception in thread "main" java.lang.RuntimeException: jssc.SerialPortException: Port name - COM3; Method name - openPort(); Exception type - Port busy.

My test code:

```
package com.arduino.model;

import org.ardulink.core.Link;
import org.ardulink.core.Pin;
import org.ardulink.core.convenience.Links;

import org.ardulink.core.linkmanager.LinkManager;
import org.ardulink.core.serial.rxtx.SerialLinkConfig;
import org.ardulink.util.URIs;

import java.util.concurrent.TimeUnit;

public class Ardulink {
    SerialLinkConfig config;
    LinkManager.Configurer configurer;
    Link link;

    public Ardulink() {
        System.out.println("ae");
        String s = "ardulink://serial?port=COM3&baudrate=115200&pingprobe=true&proto=ardulink2&waitsecs=20";
        link = LinkManager.getInstance()
                 .getConfigurer(URIs.newURI
         (s)).newLink();
       // Link link = Links.getDefault();
      // Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3"));

        try {
            Pin.DigitalPin pin = Pin.digitalPin(11);
            boolean power = true;
            while (true) {
                System.out.println("Send power:" + power);

                link.switchDigitalPin(pin, power);

                power = !power;
                TimeUnit.SECONDS.sleep(2);
            }
            } catch(Exception e){
                e.printStackTrace();
            }
        }
    }
```

The ardulink console works fine, but my code isn't.

LucianoZu, 2017-01-09:
Have you something attached to 0 o 1 PINs on your Arduino? Have you another software already listening on COM3 (maybe Ardulink console or Arduino Serial Monitor)?

elpmoraes, 2017-01-10:
No other software open when I've tried.
I'd tested with a led connected at some ports, including 0 and 1, but seems it's made no difference when i tried without the leds or something else.

pfichtner, 2017-01-10:
Perhaps we could localize the problem if the OP would tell has which operating system he is using! ;-)

elpmoraes, 2017-01-10:
Windows 10, arduino uno :P

pfichtner, 2017-01-14:
Ah Windows! I guess I found the problem! ;-)

For localizing the problem could you please tell us: 
- Does jssc work with Ardulink Console? 
- Does Ardulink Ardulink Console works reliable? 
- Does unplug/plug help sometimes? 

@LucianoZu  I guess the problem could actual be Windows 10 related, there was a similar problem within Arduino IDE: https://github.com/arduino/Arduino/pull/4792#issuecomment-204443793

elpmoraes, 2017-01-14:
am i doing something wrong ?

elpmoraes, 2017-01-14:
Does jssc work with Ardulink Console? Yes, works fine
Does Ardulink Ardulink Console works reliable?  I found no problem so far
Does unplug/plug help sometimes? I still got "port busy" in my code. Unplugging did not solve.

I'll take a look in this thread

Thanks in advance

LucianoZu, 2017-01-16:
elpmoraes,
could you please try the following?

1. Run your code with serial connection instead of serial-jssc (this use RXTX library and you have to put right .dll in the same folder of your code)

2. Run data receiver sample both with serial and serial-jssc

Thanks

pfichtner, 2017-01-16:
I interpreted the outcommented code lines as things probed that didn't work. Am I right?

LucianoZu, 2017-01-19:
Do you mean in windows pingprobe=true doesn't work? Maybe. I usually set it to false.

pfichtner, 2017-01-20:
There is no reason why pingprobe should not work when running on windows. Additionally pingprobe is just a shortcut: The maximum time to wait for a ping response is the same time a non-pingprobe connection waits/sleeps. So after that amount of time the pingprobe connection should work even if there has no ping response been received.

elpmoraes, 2017-01-22:
> java -jar example-datareceiver-2.0.1.jar
> WARNING:  RXTX Version mismatch
>         Jar version = RXTX-2.2pre1
>         native lib Version = RXTX-2.2pre2
> Error 0x5 at ..\src\termios.c(892): Acesso negado.
> 
> Exception in thread "main" java.lang.RuntimeException: gnu.io.PortInUseException: Unknown Application
>         at org.ardulink.util.Throwables.propagate(Throwables.java:22)
>         at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:468)
>         at org.ardulink.DataReceiver.createLink(DataReceiver.java:147)
>         at org.ardulink.DataReceiver.work(DataReceiver.java:94)
>         at org.ardulink.DataReceiver.doMain(DataReceiver.java:90)
>         at org.ardulink.DataReceiver.main(DataReceiver.java:78)
> Caused by: gnu.io.PortInUseException: Unknown Application
>         at gnu.io.CommPortIdentifier.open(CommPortIdentifier.java:467)
>         at org.ardulink.core.serial.rxtx.SerialLinkFactory.serialPort(SerialLinkFactory.java:108)
>         at org.ardulink.core.serial.rxtx.SerialLinkFactory.newLink(SerialLinkFactory.java:69)
>         at org.ardulink.core.serial.rxtx.SerialLinkFactory.newLink(SerialLinkFactory.java:51)
>         at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:466)
>         ... 4 more

AND, when i've tried with serial-jssc:

> java -jar example-datareceiver-2.0.1.jar -connection ardulink://serial-jssc
> Exception in thread "main" java.lang.IllegalArgumentException: No factory registered for "serial-jssc", available names are [ardulink://serial, ardulink://proxy]
>         at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
>         at sun.reflect.NativeConstructorAccessorImpl.newInstance(Unknown Source)
>         at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
>         at java.lang.reflect.Constructor.newInstance(Unknown Source)
>         at org.ardulink.util.Optional.newException(Optional.java:159)
>         at org.ardulink.util.Optional.getOrThrow(Optional.java:152)
>         at org.ardulink.core.linkmanager.LinkManager$1.getConfigurer(LinkManager.java:540)
>         at org.ardulink.DataReceiver.createLink(DataReceiver.java:147)
>         at org.ardulink.DataReceiver.work(DataReceiver.java:94)
>         at org.ardulink.DataReceiver.doMain(DataReceiver.java:90)
>         at org.ardulink.DataReceiver.main(DataReceiver.java:78)

@pfichtner 
Yes, the outcommented lines isn't working.

elpmoraes, 2017-01-22:
i had uploaded the ArdulinkProtocol sketch before, is that right, or doesn't make diff ?