Ardulink/Ardulink-2 #23: Need help! receiving custom messages

VatsaPatel, 2017-02-08:
Hello Luciano,

I need to print rotary encoder values from arduino in java.
But cannot catch the serial values in java.

This is my Java Code:
```Java
			private RplyEvent rplyEvent2;
                         public void U_speed(int pwm) throws InterruptedException, IOException {
		            System.out.println("pwm of u = " + pwm);
		            link.sendCustomMessage("encoder");
		            rplyEvent2 = null;
		            int reply = (int) rplyEvent2.getParameterValue("encoder_reply");
		            System.out.println("it is" + reply);
		            }
```
This is my arduino code:
```C
 else if(inputString.substring(6,15) == "cust/encoder") {
          Serial.print("alp://rply/");
          Serial.print("ok?id=");
        Serial.print("0");
         Serial.print("&encoder_reply=");
          Serial.print(encoder_val);
         
        Serial.print('\n'); // End of Message
        Serial.flush();
      }
```

LucianoZu, 2017-02-09:
Hi Vatsa,
if I understand well your code, you need for a way to retrieve data from Arduino when your application requires it. You can see UniqueID example in our distribution. You can see the java code here: [](https://github.com/Ardulink/Ardulink-2/tree/master/example-uniqueid/src/main/java/org/ardulink) and the sketch here: [](https://github.com/Ardulink/Ardulink-2/tree/master/deploy-dist/rootfolder/examples/UniqueID/sketch/UniqueID)

Another way to read data from Arduino with Ardulink is registering a listener on the Link object. You can see the datareceiver example here: [](https://github.com/Ardulink/Ardulink-2/tree/master/example-datareceiver/src/main/java/org/ardulink) or study openQCM project that uses our technology. Note that openQCM uses Ardulink1 while openQCM2 uses Ardulink2.

Please let me know if you need for other info.

closed 2017-02-09 by LucianoZu

VatsaPatel, 2017-02-09:
Thank you very muc for your reply Luciano,

Actually I need to display current rpm of motor in java. I am sorry for sending you the test code. Well, I need to retrieve data contantly from arduino. I am new to ardulink and could not understand usage of 	fromDeviceMessageListener or FromDeviceMessageReply and getting parameters from it, So I used example from UniqueId and somehow could not get rawdatalistener in datareceiver example.Now,  I have set an interupt in arduino that would send new rpm values when changed.

So, basically I need to contantly receive data from 3 arduino's and simultaneously send messages for other tasks also from gui.
I am sending data from arduino like this:

```C
 Serial.print("alp://rply/");
          Serial.print("ok?id=");
        Serial.print("0");
         Serial.print("&encoder_reply=");
          Serial.print(encoder_val);
         
        Serial.print('\n'); // End of Message
        Serial.flush();
```
So, can you show me correct way to this task?