Ardulink/Ardulink-2 #26: java.lang.NullPointerException

tswapnil, 2017-06-05:
Exception in thread "main" java.lang.NullPointerException
	at org.zu.ardulink.gui.SerialConnectionPanel.getConnectionPort(SerialConnectionPanel.java:129)
	at AdbClient.runApp(AdbClient.java:60)
	at AdbClient.<init>(AdbClient.java:31)
	at AdbClient.main(AdbClient.java:36)

What are the possible reasons for this error to occur ?

I am calling the runApp() in my constructor. 
private void runApp() {
    
        // Register a RawDataListener to receive data from Arduino
        link.addRawDataListener(this);
        
        // add an action listener to the connect button
                  if(serialCOM == null || link == null){
                      System.out.println("SERIAL COM or link is null");
			return;
                  }
               
                 String comPort = serialCOM.getConnectionPort();
                String baudRateS = serialCOM.getBaudRate();
                if (comPort == null || "".equals(comPort)) {
                   System.out.println("Invalid COM PORT setted.");
                } else if (baudRateS == null || "".equals(baudRateS)) {
                    System.out.println("Invalid baud rate setted. Advice: set ");
                } else {
                    try {
                        int baudRate = Integer.parseInt(baudRateS);
                        // init the connection between Arduino and PC
                        boolean connected = link.connect(comPort, baudRate);
                        if (connected) {
			   System.out.println("Serial Arduino Connection Established");
                            //connect.setEnabled(false);
                            serialCOM.setEnabled(false);
                            //disconnect.setEnabled(true);
                        }
                    } catch (Exception ex) {
                        ex.printStackTrace();
                        String message = ex.getMessage();
                        if (message == null || message.trim().equals("")) {
                            message = "Generic Error on connection";
                        }
                        //JOptionPane.showMessageDialog(connect, message, "Error", JOptionPane.ERROR_MESSAGE);
                    }       
        // add an  action  listener to disconnect button 
         
    }}

LucianoZu, 2017-06-05:
This is an Ardulink 1 issue not an Ardulink 2 issue. Please pay attention on your Ardulink version for future requests.

Null Pointer on SerialConnectionPanel on row 129 is due to port selection lack. You should select a port before call getConnectionPort(). Maybe no ports are available too. This could happen when Link instance cannot find a valid port. Please check if RXTX library is installed and work properly on your system.

I suggest to you to migrate to Ardulink 2 here you can find serial connection based on JSSC that is simpler to use.

closed 2017-06-05 by LucianoZu

tswapnil, 2017-06-06:
Thanks LucianoZu, 
   I have tried to get the port List from - "List<String> portList = link.getPortList();" but could not find any port . link is the default instance.
 
I also  tried to work with ardulink 2 .  I ran the console (I am using ubuntu machine ) (java -jar ardulink-console-2.0.1.jar)  .  On the widget too , it does not show any port . I have connected adafruit's flora and am able to see the port on arduino as well .  I have also tried to manually assign the port - "/dev/ttyACM1" - I get a NoSuchPortException.     
   Could you suggest anything for ardulink2 or ardulink 1 that helps me connect to the serial port. 
Thanks

LucianoZu, 2017-06-06:
Ok,
the problem is RXTX that cannot detect ports called /dev/ttyACM[x] on Linux systems. Have you tried with Ardulink 2 console and JSSC (not RXTX) serial link?

However you should solve the problem with a linux symlink. I had the same problem with my Raspberry. Check it here:
http://www.ardulink.org/how-to-install-arduino-pc-link-on-raspberry-pi/

This is the content you are interested in:
"Unfortunately RXTX does not handle devices like /dev/ttyACM… but Raspberry PI sees Arduino usb connection just like a serial port of this kind. In other words Ardulink can not find the door to establish the connection. To fix the problem, you still need a symbolic link. Type the command:
ln -s /dev/ttyACM[x] /dev/ttyS8[x]
Where x should be a number. Probably zero. For more details see:
http://pblog.ebaker.me.uk/2011/09/processing-usb-ports-devttyacm0.html"

Thanks to you and enjoy using Ardulink.

tswapnil, 2017-06-06:
Thanks . That was really helpful. But another error keeps on coming now -

Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
Jun 06, 2017 2:14:54 AM org.zu.ardulink.ConnectionContactImpl writeLog
INFO: found the following ports:
Jun 06, 2017 2:14:54 AM org.zu.ardulink.ConnectionContactImpl writeLog
INFO:    /dev/ttyS85
Jun 06, 2017 2:14:54 AM org.zu.ardulink.ConnectionContactImpl writeLog
INFO:    /dev/ttyS80
PortList Size 2
/dev/ttyS85
/dev/ttyS80
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f4786d3c462, pid=9388, tid=0x00007f4786c34700
#
# JRE version: Java(TM) SE Runtime Environment (8.0_131-b11) (build 1.8.0_131-b11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.131-b11 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [librxtxSerial.so+0x6462]  read_byte_array+0x52
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/swapnil/Robotics/Server/hs_err_pid9388.log
Jun 06, 2017 2:14:54 AM org.zu.ardulink.ConnectionContactImpl writeLog
INFO: connection on /dev/ttyS80 established
Serial Arduino Connection Established
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted (core dumped)

This error comes after I run both ardulink 1 and ardulink 2 . Ardulink 2 finds ports after making symlinks. As soon as I click on connect button . It shows the above error. 
It would help if you could tell how to resolve this . Did you come across this issue ? 

Thanks

LucianoZu, 2017-06-06:
Unfortunately RXTX is not maintained anymore so the problem could be generated by an incompatibility issue between your SO and RXTX library. So you should search on the web for it.

Do you have a similar problem with JSSC on Ardulink 2? I suggest you to migrate to this Link implementation that is easier to use and  more stable. Of course it is available in Ardulink 2 only if you prefer Ardulink 1 you should implement a JSSC connection for it.