Ardulink/Ardulink-2 #27: MQTT Authorization support

keyndark, 2017-08-05:
Hi. I'm not found in docs how to pass authorization credentials for MQTT server. Is this feature supported or any plan to implement it?

pfichtner, 2017-08-05:
Adding authorization is not a great thing and could be added without any problem. In which component are you missing it? MqttMain? ArdulinkConsole?

pfichtner, 2017-08-05:
MqttMain in git repo now supporting authentification

pfichtner, 2017-08-05:
MqttLink in git repo now supporting authentification

keyndark, 2017-08-05:
Great! Thank you!

pfichtner, 2017-08-05:
Since for now there's now way for downloading the artifacts of Ardulink's CI build you'll have to pull and build by yourself or to wait for the next Ardulink release

pfichtner, 2017-08-05:
@LucianoZu Please close when releasing

pfichtner, 2018-02-25:
Enhancement included in release v2.1.0 Gordio

closed 2018-02-25 by pfichtner