Ardulink/Ardulink-2 #30: cant find ardulink.jar file

branewyn, 2017-09-26:
hi. new to the ardulink scene. very new to programming. 
i am working with java and would like to build a command line interface using ardulink. i have opened the ardulink console and it is communicating to my arduino which means everything is set up correctly on my computer and the arduino sketch is working.
i cannot however seem to get it working from a java perspective. i do not want the GUI. just text input. i cannot find anything on the web that can help me. the trouble that i have is importing the ardulink jar files. i do not know which one or ones i should use. i tried importing the entire lib folder into eclipse and pasted all the jar files in the lib folder where my source program is. still no joy
any help would be appreciated

pfichtner, 2017-09-26:
Did you add the added jars to the classpath?

branewyn, 2017-09-26:
I did not. How do I do that and which jars do I use?

branewyn, 2017-09-26:
Netbeans. I can load jar files. That is not the problem. The problem is which jar should be loaded and where should it be loaded? I have ardulink 2

pfichtner, 2017-09-26:
There is no point where you load a jar file. You have to add them to your classpath. From the perspective of playing around with Ardulink there is no reason why you should not add all the jar files to the classpath.

branewyn, 2017-09-26:
Okay.i will do that. Thanks.

branewyn, 2017-09-26:
added the jar path to the variable. the next bit im not sure about and i cant find any examples. that is what is annoying and you seem to be the only one that is replying.
i have ardulink 2.0.1 loaded onto my PC. 
the first part is that the classes are not called org.zu.ardulink.<something>. 
they are org.ardulink.core.<something> 
the second part is that in the examples that i see online is that they import an ardulink.jar file.
i cannot find this jar file in my downloads.
i only have ardulink-camel-2.0.1,ardulink-console-2.0.1(which is that gui application),ardulink-core-base-2.0.1(which contains a Link.class file),ardulink-core-beans-2.0.1(not sure what this is)
so going through the class i can see that the link class file is somewhat what i needs. but running the getDefaultInstance() breaks my program

pfichtner, 2017-09-26:
There are a bunch of examples inside the project's root, these modules start with "example-" e.g. example-datareceiver. Please see https://github.com/Ardulink/Ardulink-2/blob/master/example-datareceiver/src/main/java/org/ardulink/DataReceiver.java for an example of the imports. You should just checkout/clone the whole repository and modify DataReceiver.java to fulfill your needs.

pfichtner, 2017-09-26:
btw: I am the only one responding because this is an issue tracker for bugs and feature requests and not a discussion forum.

branewyn, 2017-09-26:
i see. thanks. where do i find the forum? on the web page? thanks for the link. i will go check it out

branewyn, 2017-09-26:
hi again

so what im trying to do is blink led 13 on my board. nothing fancy.

do you guys have basic java files that you guys use to do testing.
please note that i am very new to any of this. im only trying to do very basic things.
all the examples are very complicated and excessive and well beyond my scope currently.
do you also perhaps have some basic form of documentation as to how to work with all these classes?
ie. if you want to do this thing. you need all these things. and that is how you use it to get it working.
basic guidelines as it were?
because right now i dont understand half of what im doing and that defeats the purpose of it all.

this is where i am currently.
i have loaded these jar files
core-base
core-serial-rxtx
core-util

i am using windows 7. 
coding on netbeans. 
i have an arduino uno and have loaded the ArdulinkProtocol.ino file successfully on the arduino.
these are the imports that i have done

import java.io.IOException;
import java.util.List;
import java.util.Scanner;
import java.util.concurrent.TimeUnit;
import org.ardulink.core.Link;
import org.ardulink.core.Pin;
import org.ardulink.core.Pin.DigitalPin;
import org.ardulink.core.convenience.Links;

import org.ardulink.core.linkmanager.LinkManager;
import org.ardulink.core.serial.rxtx.SerialLinkConfig;
import org.ardulink.util.URIs;

 i have found the below code and am running this.

public static void main(String[] args) throws InterruptedException, IOException {
       SerialLinkConfig config;
       LinkManager.Configurer configurer; 
       Link link;// = Links.getDefault();
        String s = "ardulink://serial?port=COM13&baudrate=115200&pingprobe=true&proto=ardulink2&waitsecs=20";
        link = LinkManager.getInstance().getConfigurer(URIs.newURI(s)).newLink();
        try {
            Pin.DigitalPin pin = Pin.digitalPin(13);
            boolean power = true;
            while (true) {
                System.out.println("Send power:" + power);

                link.switchDigitalPin(pin, power);

                power = !power;
                TimeUnit.SECONDS.sleep(2);
            }
            } catch(Exception e){
                e.printStackTrace();
            }
        }

this is the error that i am getting

Exception in thread "main" java.lang.NoClassDefFoundError: org/ardulink/core/beans/BeanProperties
	at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.<init>(LinkManager.java:366)
	at org.ardulink.core.linkmanager.LinkManager$1.getConfigurer(LinkManager.java:544)
	at blinkled.BlinkLed.main(BlinkLed.java:27)
Caused by: java.lang.ClassNotFoundException: org.ardulink.core.beans.BeanProperties
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 3 more
C:\Users\228525\AppData\Local\NetBeans\Cache\8.2\executor-snippets\run.xml:53: Java returned: 1
BUILD FAILED (total time: 0 seconds)

please help as i do not know how to fix any of this?

pfichtner, 2017-09-26:
- THIS IS NOT A HELP FORUM HERE!
- Why are you asking what is **ardulink-core-beans**-2.0.1 for, don't add it to the classpath? What do you think could be wrong when not adding this jar to your classpath and java ist telling you that the a class named  **org.ardulink.core.beans**.BeanProperties is missing? 
- Why don't you just follow the instructions you got and add **all** the jars to your classpath? 
- Why don't you just use the simple sample from the main page https://github.com/Ardulink/Ardulink-2? 
- THIS IS NOT A HELP FORUM HERE!

LucianoZu, 2017-09-27:
Hi  Branewyn,

you can ask for here  [http://www.ardulink.org/contact-us/](http://www.ardulink.org/contact-us/)

branewyn, 2017-09-27:
please dont be mean. while you may have years of java experience and know everything there is to know i have 2 and a half months. i dont know a fraction of what im supposed to know or how the error reporting services in java work. the only errors i have encountered thus far is syntax and logic errors. this is my first venture beyond pure java. whether i ask here or on the ardulink page makes no difference as both gets the job done. and the web page also says that i can ask questions here or there. but thanks. i get that you want to keep the two seperate. also. i got it working thanks. also got a binding error for the log file. but i will sort that out myself as the error reporting is actually useful in telling you what to do and where to go

pfichtner, 2017-09-28:
As you described above you're a Java beginner. There is nothing wrong with that and I hope you get a lot of help from the community. But you mixed Ardulink's issue and feature tracker with a java beginners help forum. 

Could you add Ardulink's jars to the classpath: No you coluldn't
Could you add any other jars to the classpath: No you coluldn't
So is this an Ardulink problem or is this even Ardulink related: No, it isn't! 

So yes: You can ask here for Ardulink questions, problems, issues and feature requests. 

If your new to card driving and you cannot open the car's door because you have no clue about cars, doors or even keys. Who would you ask for help to get the door opened? The engineers of the model of your car?

branewyn, 2017-09-28:
Thank you peter. I will ask it on the correct forum.
But just as a reply. Please note that I did say that I added jar files (specifically the rxtx library) already. Ie I can add jar files. I did however not know what you referred to when you said "add the jars to the class path". To you...and advanced java programmer that easily means import the jars as that makes sense...but when you say it to a beginner..it could just as easily mean to add the path of the jars to the path variable in the windows settings...please take note that not everyone is on your level of understanding.

Regarding the car door thing....if it was a fancy car with weird opening doors that you have no reference to or instructions on how to open then yes....i would ask the engineers of the door on how to open it.

Just like this is a platform to use a device in a way that it was not designed to do. I mean...who else would you ask on how to use it?

But thanks. I will direct my questions to the correct forum.

pfichtner, 2017-09-28:
Ardulink is no fancy car (jar), it's jar like any other. 
And no, your would not ask the car's engineers (with cars you even wouldn't know them): You would ask other drivers or perhaps the local dealer.

branewyn, 2017-09-28:
Sure it is. And very much so. If you considered it a lame project you wouldnt spend your time and energy on it. So yes. Its potential and ability is mind blowing. It is why im here in the first place.

Also...to you...an experienced coder...it is easy to load everything....as you have loaded countless of jars....to me...again...a beginner...has loaded 1 before this...rxtx...when you told me to load all the jars...my first response wasnt sure lets do it....it was what? You can load more than one? That is crazy.