Ardulink/Ardulink-2 #32: Mention Stackoverflow tag in README.md

pfichtner, 2017-09-26:
We should offer a way for Ardulink users to discuss things. Stackoverflow provides such a functionality. There is at least one question with tag "ardulink" and I did subscribe on that tag. 

If you agree you should add a note in README.md like "We are listening on #ardulink at stackoverflow". If you prefer some other platforms please refer them or just close the issue if you think there is no need.

pfichtner, 2017-09-26:
I have to correct myself: Stackoverflow is not a discussion forum but a Q&A platform. If we would like to have a discussion forum perhaps we could use slack or similar things for that.

LucianoZu, 2017-09-27:
Stackoverflow is a good platform, I've replied to some question a long time
ago. So we could use it, I agree.
Unfortunately may be I have to do another account because actually on
stackoverflow I haven't a good reputation (when I started after one or two
replies someone has voted -1 to me). I have a good reputation on
https://arduino.stackexchange.com

Luciano

   - Profilo LinkedIn: http://www.linkedin.com/in/lucianozu
   - Ardulink: http://www.ardulink.org/
   - Lucio Caio Autore: http://luciocaio.altervista.org/

JuniorEC, 2017-10-25:
Hello, I have a sugentão, I'm starting the Ardulink library studies, and at first I felt deep difficulties to use the methods in general. They could make a few basic examples with the new version for beginners of library studies.

p.s - some examples of the sites, are outdated and did not work for me.

I liked the idea of ​​stackoverflow, congratulations to the initiative.