Ardulink/Ardulink-2 #33: Error on Ardulink Console

JuniorEC, 2017-10-30:
[image]

During the execution of the Ardulink Console, the attached error appeared. Is something missing to run?

LucianoZu, 2017-10-31:
Please read the stack trace log on Java console.

pfichtner, 2017-10-31:
However the OP did accomplished that but that's the message you get when deleting ardulink-core-beans.jar from the lib folder. I guess the stacktrace tells ```java.lang.NoClassDefFoundError: org/ardulink/core/beans/BeanProperties```

pfichtner, 2017-10-31:
To make life easier in future I added the exception's class name to the JPanel and the error dialog with commit 8d3a96b7ffb12cc502eb09795f41350efd4049dc

closed 2020-05-20 by pfichtner