Ardulink/Ardulink-2 #37: Does Ardulink-2 support Arduino Mega2560 board/Onewire

ilovetogetspamed, 2018-06-05:
Hi, does your code work on the 2560 or just UNOs?  How about Onewire support?  Built in or would I have to kludge it in somehow?
Thanks

LucianoZu, 2018-06-06:
Ardulink works with a lot of devices. It has a several Link implementations (Serial, Bluetooth, MQTT, specific ones for instance for Raspberry and so on) over some of this links and more precisely over stream links there are communication protocols. For instance there is the Ardulink protocol used for Arduino boards but also a protocol for NodeMCU/LUA boards.

So Arduino Mega works with Ardulink just like UNO. About Onewire, [Maxim Integrated](https://www.maximintegrated.com/en/products/digital/one-wire/software-tools/api-java-sdk.html) supports java so should be pretty easy to develop an Ardulink Link implementation for this boards.

ilovetogetspamed, 2018-06-06:
Thank you.  I appreciate your time.
Chris