Ardulink/Ardulink-2 #38: Can`t read data sent from arduinio in java.

kushalsha133, 2018-06-22:
`package arduino1;
import com.fazecast.jSerialComm.*;

import java.util.Scanner;

import javax.swing.*;

public class arduino {

	public static void main(String[] args) {
		// TODO Auto-generated method stub
		/*JFrame wind=new JFrame();
		JSlider slider= new JSlider();
		slider.setMaximum(1023);
		wind.add(slider);
		wind.pack();
		wind.setVisible(true);*/
		//SerialComm ports[] = SerialComm.getCommPorts();
		SerialPort ports[] = SerialPort.getCommPorts();
		
		int i=1;
		System.out.println("Available Ports");
		for(SerialPort port : ports)
		{
			System.out.println(i++ +": "+port.getSystemPortName());
		}
		Scanner sc =new Scanner(System.in);
		System.out.println("Select a port\n");
		int choosenPort;
		choosenPort = sc.nextInt();
		
		SerialPort port= ports[choosenPort-1];
		try
		{
			port.openPort();
			System.out.println("Successfully opened port");
		}catch(Exception e){
			System.out.println(e);
		}
		
		port.setComPortTimeouts(SerialPort.TIMEOUT_READ_SEMI_BLOCKING, 0, 0);

	
		Scanner data =new Scanner(port.getInputStream());  //error line
		int number=0;
		while(data.hasNextLine())
		{
			try{number=Integer.parseInt(data.nextLine());}catch(Exception e) {}
			System.out.println(number);
		}	
	}
}`

error:
Available Ports
1: COM3
Select a port

1
Successfully opened port
Exception in thread "main" java.lang.NullPointerException
	at java.base/java.io.Reader.<init>(Unknown Source)
	at java.base/java.io.InputStreamReader.<init>(Unknown Source)
	at java.base/java.util.Scanner.<init>(Unknown Source)
	at arduino1.arduino.main(arduino.java:44)

It showed output once but has been giving this error ever since then. I need to have the data from the 6 sensors i have connected to the arduino which are sending digital data 0/1.

I am new to github please comment if this way of writing, asking, posting is wrong.

LucianoZu, 2018-06-22:
Hi kushalsha133,
you are trying to use java to open a serial port with a library I don't know. So you should open an issue to that library.

Ardulink is a different project that use several ways to connect computers to hardware boards (arduino and more).

If you want use Ardulink to collect data from arduino you can please read our examples code or articles on our site.

closed 2018-06-22 by LucianoZu

pfichtner, 2018-06-22:
You should use Ardulink instead of reimplementing the wheel. Ardulink encapsulates the whole complexity you're stumbling across. And if you're using Ardulink and have problems whith Ardulink you can raise an issue at Ardulink ;-)