Ardulink/Ardulink-2 #39: NullPointerException when connecting via network proxy server

jpink, 2018-10-14:
Hi!

First I had problems with TRTX library on Linux. Library files (.so) aren't included in the lib folder. Easiest way is install them using `apt-get install librxtx-java` in Debian or Ubuntu but the version **2.2pre2** might be too new because it expects **2.1-7**. Then I found the right version in _"3rds\rxtx-2.1-7-bins-r2.zip\rxtx-2.1-7-bins-r2\Linux\i686-unknown-linux-gnu"_ folder. But then I need to tell its path: `java -Djava.library.path=/var/ardulink/lib -jar ardulink-networkproxyserver-2.1.0.jar start` on start script.

Second problem is the **NullPointerException** when trying to connect Arduino. I can connect to the proxy and list ports. The only port that i have is **/dev/ttyUSB0**. When i click **Connect** I get the following stacktrace:

> Ardulink Network Proxy Server running...
OpenJDK Server VM warning: You have loaded library /var/ardulink/lib/librxtxSerial.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
Oct 14, 2018 9:15:11 PM org.ardulink.core.StreamReader readUntilClosed
SEVERE: Error while Reader Initialization
java.lang.NullPointerException
        at org.ardulink.core.StreamReader.readUntilClosed(StreamReader.java:68)
        at org.ardulink.connection.proxy.NetworkProxyServerConnection.run(NetworkProxyServerConnection.java:96)
        at java.lang.Thread.run(Thread.java:748)
> 
> Oct 14, 2018 9:15:11 PM org.ardulink.connection.proxy.NetworkProxyServerConnection run
SEVERE: Error while doing proxy
java.lang.NullPointerException
        at org.ardulink.core.StreamReader.readUntilClosed(StreamReader.java:83)
        at org.ardulink.connection.proxy.NetworkProxyServerConnection.run(NetworkProxyServerConnection.java:96)
        at java.lang.Thread.run(Thread.java:748)
> 
> Oct 14, 2018 9:15:12 PM org.ardulink.connection.proxy.NetworkProxyServerConnection run
INFO: /192.168.1.55:55678 connection closed.

I seems that scanner is null when calling that:
https://github.com/Ardulink/Ardulink-2/blob/7b03289368c19437b2959ecf87579c32af67cd53/ardulink-networkproxyserver/src/main/java/org/ardulink/connection/proxy/NetworkProxyServerConnection.java#L96

StreamScanner isn't initialized on constructor so it is null when calling readUntilClosed. Maybe the proxy misses calling runReaderThread or someting.

LucianoZu, 2018-10-15:
Hi,
rxtx was at v2.1.7 for years. I didn't know they have a new version. I'll try to upgrade it in our pom.xml dependencies list.
However you can try to use jssc link that is another usb serial link. It uses jssc library instead of rxtx and all shared objects needed for it are inside the jar.

About the null pointer Peter and I will check this as soon as possible.

Thank you
   Luciano

jpink, 2018-10-17:
@LucianoZu: Thanks for the rxtx update! That will help installing ardulink-proxy as a daemon in linux. I can share (when i got it working) my init.d script if someone wants that proxy to installed as a service and started on (re)boot.

@pfichtner: I was trying on version 2.1.0. Now I will try it on 2.0.1 because that NetworkProxyServer isn't changed for years so maybe latest core changes has broke it. Or is it working on Windows?

pfichtner, 2018-10-18:
NetworkProxyServer read thread problem fixed. After fixing I realized that there is a  inherited error. NetworkProxyServer unfortunately is the only burden of the past which therefore has no tests. My hope was that it will be replaced by MQTT by users which not seems to be the case.

pfichtner, 2018-10-27:
Anyone still interested? So please review otherwise I will close this issue without review.

pfichtner, 2018-11-02:
Nobody interessted in review so closing as fixed

closed 2018-11-02 by pfichtner