Ardulink/Ardulink-2 #45: URISyntaxException Thrown while Calling Links.getDefault()

rajb865, 2019-04-25:
Hi As per the description given at [ArduinoLink2-Github](https://github.com/Ardulink/Ardulink-2), I am following the same way to control the Arduino LED.

I have added dependencies to my `pom.xml` as below

`<dependencies>
		<dependency>
			<groupId>org.ardulink</groupId>
			<artifactId>ardulink-core-base</artifactId>
			<version>2.1.0</version>
		</dependency>
		<dependency>
			<groupId>org.ardulink</groupId>
			<artifactId>ardulink-core-serial-jssc</artifactId>
			<version>2.1.0</version>
		</dependency>
	</dependencies>`

I am having one sample java application which is calling the below line from `main()` method 
`Link link = Links.getDefault();`

but when I run this HelloWorld program I am getting this error

`java.lang.RuntimeException: java.net.URISyntaxException: Illegal character in path at index 7: wh�what`

Please let us know why is this happening? How can I fix it?

rajb865, 2019-04-25:
Somewhere on the internet I found the below solution to retrieve the Link object

```
Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3&baudrate=9600&pingprobe=false&waitsecs=1"));
```

with above solution I am not getting the `URISyntaxException`.  is this the valid solution to retrieve the `Link` object? with the above solution I am not able to change the state of the LED. here is the code snippet

```
Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3&baudrate=9600&pingprobe=false&waitsecs=1"));

		DigitalPin pin = Pin.digitalPin(13); 
		link.switchDigitalPin(pin, true);
```

So please let me know if this is the right way to change the state of `LED 13`?

rajb865, 2019-04-25:
I have uploaded [ArdulinkProtocol.ino](https://github.com/Ardulink/Ardulink-2/blob/master/deploy-dist/rootfolder/sketches/ArdulinkProtocol/ArdulinkProtocol.ino) sketch to the `Arduino UNO` board. I tried running java application again but I did not get any luck.

pfichtner, 2019-04-26:
I do not exactly understand what you do or what you expect from Ardulink. Without the Ardulink sketch on the connect Arduino your java code never will work. 

As you can see in the sketch you have linked, the sketch sets the baudrate to 115200 so trying to communicate with a baudrate of 9600 never will work and also if your comport is COM3 is questionable

As you can see in https://github.com/Ardulink/Ardulink-2/blob/69a2d0953d1eecc10df005fd4dbfac2c2800be50/ardulink-core-base/src/main/java/org/ardulink/core/convenience/Links.java#L66  ```Link link = Links.getDefault();``` uses ```ardulink://serial```. So does ```Link link = Links.getLink(URIs.newURI("ardulink://serial"));``` work?

rajb865, 2019-04-26:
Hi @pfichtner ,

Thanks for your reply. Using the below code I am able to blink the LED.

```
public static void main(String[] args) throws IOException, InterruptedException {
		
		Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3&baudrate=115200&pingprobe=false&waitsecs=1"));
		DigitalPin pin = Pin.digitalPin(13); 
		boolean state = false;
		while (true) {
			if (state) {
				link.switchDigitalPin(pin, false);
				state = false;
			}else {
				link.switchDigitalPin(pin, true);
				state = true;
			}
			Thread.sleep(1000);
		}
}
```
But if I simply turn **ON/OFF** the LED that does not work. below is the sample code for that

```
public static void main(String[] args) throws IOException, InterruptedException {
		
		Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3&baudrate=115200&pingprobe=false&waitsecs=1"));
		DigitalPin pin = Pin.digitalPin(13); 
		link.switchDigitalPin(pin, true);
}
```

I am using the same [ArdulinkProtocol.ino](https://github.com/Ardulink/Ardulink-2/blob/master/deploy-dist/rootfolder/sketches/ArdulinkProtocol/ArdulinkProtocol.ino) sketch at Arduino board.

could you please help me why simple **ON/OFF** is not working?

pfichtner, 2019-04-26:
Could you please try what I did suggest instead of trying things? 
```
		Link link = Links.getLink(URIs.newURI("ardulink://serial"));
		DigitalPin pin = Pin.digitalPin(13); 
		link.switchDigitalPin(pin, true);
}```

rajb865, 2019-04-29:
Hi @pfichtner ,

I tried with the code you provided and got the below exception.

```
Exception in thread "main" java.lang.IllegalArgumentException: No factory registered for "serial", available names are [ardulink://serial-jssc]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.ardulink.util.Optional.newException(Optional.java:158)
	at org.ardulink.util.Optional.getOrThrow(Optional.java:151)
	at org.ardulink.core.linkmanager.LinkManager$1.getConfigurer(LinkManager.java:603)
	at org.ardulink.core.convenience.Links.getLink(Links.java:93)
	at com.dover.arduino.ArduinoMain.main(ArduinoMain.java:30)
```

Please let me know what should I have in place to correct this.

LucianoZu, 2019-04-29:
..hhhmmm... Ardulink has a discovery mechanism to all knew Link implementations. So you have just to add the right JAR into the classpath in order to allow Ardulink register it. Have you added serial-jssc to classpath? Please try this.

Thanks

rajb865, 2019-04-29:
Hi @LucianoZu ,

I have added dependencies through pom file. below is the code

```
<dependencies>
		<dependency>
			<groupId>org.ardulink</groupId>
			<artifactId>ardulink-core-base</artifactId>
			<version>2.1.0</version>
		</dependency>
		<dependency>
			<groupId>org.ardulink</groupId>
			<artifactId>ardulink-core-serial-jssc</artifactId>
			<version>2.1.0</version>
		</dependency>
	</dependencies>
```

Let me know if I need to explicitly add it.

pfichtner, 2019-04-29:
@rajb865 

I told you
```...if ardulink://serial won't work try ardulink://serial-jssc (depends on the Ardulink version)```
You tried ```serial```
Now Ardulink complains ```No factory registered for "serial", available names are [ardulink://serial-jssc]```
Did you try to replace ```serial``` by ```serial-jssc``` as I told you?

rajb865, 2019-04-29:
Hi @pfichtner ,

Yes I tried that too and got the below error

```
Exception in thread "main" java.lang.RuntimeException: jssc.SerialPortException: Port name - null; Method name - openPort(); Exception type - Null not permitted.
	at org.ardulink.util.Throwables.propagate(Throwables.java:42)
	at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:536)
	at org.ardulink.core.convenience.Links.getLink(Links.java:108)
	at org.ardulink.core.convenience.Links.getLink(Links.java:92)
	at com.dover.arduino.ArduinoMain.main(ArduinoMain.java:29)
Caused by: jssc.SerialPortException: Port name - null; Method name - openPort(); Exception type - Null not permitted.
	at jssc.SerialPort.openPort(SerialPort.java:161)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.serialPort(SerialLinkFactory.java:102)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(SerialLinkFactory.java:57)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(SerialLinkFactory.java:46)
	at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:534)
	... 3 more
```

Please let me know if I am missing something.

rajb865, 2019-04-29:
Hi @pfichtner ,

I tried below code and it just worked perfectly

```
Link link = Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3"));
DigitalPin pin = Pin.digitalPin(13); 
link.switchDigitalPin(pin, true);
```

Thank you so much for all the help and support you provided.

pfichtner, 2019-04-29:
That's what I wanted to achieve step by step. So nice it's working. 
On the other hand I barely can understand the problem you got with the ```Links``` class. 

I did download the JAR from maven central everything looks ok, the decompiled Links class reads
```
64 List<URI> availableURIs = linkManager.listURIs();
65 URI serial = URIs.newURI("ardulink://serial");
66 if (availableURIs.contains(serial)) {
```
Is your JAR corrupted? Please check the md5sum
```0c0c53a991eab6c49615026c9067ee59  ardulink-core-base-2.1.0.jar```

rajb865, 2019-04-29:
Jar file was right but it was me who was having very minimal information about Ardulink library. once you corrected my mistakes it worked. 

Now I will explore it more with some other examples. Thank you very much again to you Peter.

pfichtner, 2019-04-29:
This has nothing to do with your experience regarding Ardulink: Links#getDefault throws an exception with a strange error message. 
I'd like to reopen this issue since the problem still exists, we just found an alternative way to get it run by calling the same code as Links#getDefault does. 

Could you please retry Links#getDefault and send the compete stacktrace?

rajb865, 2019-04-29:
Hi Peter,

Please find logs below

```
Exception in thread "main" java.lang.IllegalStateException: Waited for arduino to boot but no response received
	at org.ardulink.util.Preconditions.checkState(Preconditions.java:48)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.waitForArdulink(SerialLinkFactory.java:87)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(SerialLinkFactory.java:70)
	at org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(SerialLinkFactory.java:46)
	at org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink(LinkManager.java:534)
	at org.ardulink.core.convenience.Links.getLink(Links.java:108)
	at org.ardulink.core.convenience.Links.getDefault(Links.java:56)
	at com.dover.arduino.ArduinoMain.setupArduino(ArduinoMain.java:68)
	at com.dover.arduino.ArduinoMain.main(ArduinoMain.java:34)
```

pfichtner, 2019-04-29:
This is not the exception originally got, as you can see we went successfully through Links#getDefault. Is this reproducible? Is ```Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3"))``` working?

pfichtner, 2019-04-29:
An explanation for the last error could be, that you have more than one COM port and the Arduino is not connected to the first one (which #getDefault assumes). 
But this is not an explanation for the originally error. But if this error no longer occurs we could close the issue.

rajb865, 2019-05-02:
Hi Peter,

`Links.getLink(URIs.newURI("ardulink://serial-jssc?port=COM3"))` is working perfectly for me. 

about the error, I am getting only above error. So We can close this issue.

pfichtner, 2019-05-02:
The issue is about that ```Links#getDefault``` is not working. 

Perhaps ```getDefault``` will not work for you because you have some other devices and ```getDefault``` uses  the first one but I'd like to know what weird exception you originally got (```Illegal character in path at index 7: wh�what```). So if you still get this error? If yes we need further investigations on it if not we can close the issue.

rajb865, 2019-05-02:
Ohh I remember Peter. I am not getting this exception now. We can close this issue now.

pfichtner, 2019-05-02:
Not very satisfying but the originally exception has been gone. Hope it never will come back ;-)

closed 2019-05-02 by pfichtner

pfichtner, 2019-05-02:
@rajb865 for your reference (this should work for you)

How to use the default link but change some attributes
```
Configurer configurer = Links.getDefaultConfigurer();
configurer.getAttribute("port").setValue("COM3");
Link link = configurer.newLink();
```

@LucianoZu Should we add this info to the README.md or anywhere else?