Ardulink/Ardulink-2 #46: eclipse palettes

donatog, 2019-04-27:
Hi!
Is it possible to add inside normal palettes into general swing components, ardulink palettes components? 
example ad-hoc: a JFrame connectionPanel or a buttonConnect, SignalButton etc..
In such a way it'd be more easy to make a mask for whom is a very beginner programmer.

LucianoZu, 2019-04-27:
Hi,
I don't understand exactly what you mean. Ardulink already has a set of swing components. Ardulink console is just an application that compose all these components.

donatog, 2019-04-28:
I was meaning in the design mode of a window application.
Eclipse, or other IDEs, have a set of palettes that can be dragged on the window frame when you are building a window application.

closed 2019-04-28 by donatog