Ardulink/Ardulink-2 #49: jSSC and RXTX crashing JVM in Windows 10

VTHMgNPipola, 2019-05-13:
I was running Ardulink with jSSC just fine in Elementary OS Juno, but now that I'm trying to run it in Windows 10, it crashes in both jSSC and in RXTX, with EXCEPTION_ACCESS_VIOLATION.

Here is the full log:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=2928, tid=940
#
# JRE version: Java(TM) SE Runtime Environment (12.0+33) (build 12+33)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (12+33, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -javaagent:D:\Users\EIGHT\Documents\ideaC.2019.1\lib\idea_rt.jar=50160:D:\Users\EIGHT\Documents\ideaC.2019.1\bin -Dfile.encoding=UTF-8 com.kerbodyne.rocketcontroller.KerbodyneRocketController

Host: Intel(R) Core(TM) i3-3220 CPU @ 3.30GHz, 4 cores, 7G,  Windows 8.1 , 64 bit Build 9600 (6.3.9600.17415)
Time: Mon May 13 08:58:03 2019 Hora oficial do Brasil elapsed time: 1 seconds (0d 0h 0m 1s)

---------------  T H R E A D  ---------------

Current thread (0x000000e3ae3a4800):  JavaThread "main" [_thread_in_native, id=940, stack(0x000000e3ada50000,0x000000e3adb50000)]

Stack: [0x000000e3ada50000,0x000000e3adb50000],  sp=0x000000e3adb4ee30,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  org.ardulink.core.serial.jssc.SerialLinkFactory.serialPort(Lorg/ardulink/core/serial/jssc/SerialLinkConfig;Ljava/lang/String;)Ljssc/SerialPort;+10
j  org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(Lorg/ardulink/core/serial/jssc/SerialLinkConfig;)Lorg/ardulink/core/convenience/LinkDelegate;+8
j  org.ardulink.core.serial.jssc.SerialLinkFactory.newLink(Lorg/ardulink/core/linkmanager/LinkConfig;)Lorg/ardulink/core/Link;+5
j  org.ardulink.core.linkmanager.LinkManager$DefaultConfigurer.newLink()Lorg/ardulink/core/Link;+12
j  org.ardulink.core.convenience.Links.getLink(Lorg/ardulink/core/linkmanager/LinkManager$Configurer;)Lorg/ardulink/core/Link;+40
j  org.ardulink.core.convenience.Links.getDefault()Lorg/ardulink/core/Link;+3
j  com.kerbodyne.kerbolang.comms.ArdulinkCommunotron.<init>()V+5
j  com.kerbodyne.rocketcontroller.KerbodyneRocketController.reconnectArduino()V+4
j  com.kerbodyne.rocketcontroller.KerbodyneRocketController.main([Ljava/lang/String;)V+11
v  ~StubRoutines::call_stub

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0xffffffffac32e3e3

Register to memory mapping:

RIP=0x000000007110b5db jSSC-2.8_x86_64.dll
RAX=0xffffffffac32e3e3 is an unknown value
RBX=0x000000e3adb4eea0 is pointing into the stack for thread: 0x000000e3ae3a4800
RCX=0x00007ffc3f750d3a ntdll.dll
RDX=0x0 is NULL
RSP=0x000000e3adb4ee30 is pointing into the stack for thread: 0x000000e3ae3a4800
RBP=0x000000e3adb4efb0 is pointing into the stack for thread: 0x000000e3ae3a4800
RSI=0x000000e3ae3a4b30 points into unknown readable memory: e0 1d 3b 24 fc 7f 00 00
RDI=0x0000000000000484 is an unknown value
R8 =0x000000e3adb4ed28 is pointing into the stack for thread: 0x000000e3ae3a4800
R9 =0x000000e3adb4ee20 is pointing into the stack for thread: 0x000000e3ae3a4800
R10=0x0 is NULL
R11=0x0000000000000213 is an unknown value
R12=0x000000e3adb4f068 is pointing into the stack for thread: 0x000000e3ae3a4800
R13=0x000000e3cea4f640 points into unknown readable memory: 44 00 3a 00 2e 00 00 00
[... 413 lines cut ...]
Log output configuration:
 #0: stdout all=warning uptime,level,tags
 #1: stderr all=off uptime,level,tags

Environment Variables:
PATH=C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin
USERNAME=EIGHT
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 8.1 , 64 bit Build 9600 (6.3.9600.17415)

CPU:total 4 (initial active 4) (2 cores per cpu, 2 threads per core) family 6 model 58 stepping 9, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, clmul, erms, ht, tsc, tscinvbit, tscinv

Memory: 4k page, system-wide physical 7878M (3665M free)
TotalPageFile size 9094M (AvailPageFile size 4480M)
current process WorkingSet (physical memory assigned to process): 69M, peak: 69M
current process commit charge ("private bytes"): 209M, peak: 211M

vm_info: Java HotSpot(TM) 64-Bit Server VM (12+33) for windows-amd64 JRE (12+33), built on Feb 21 2019 02:31:50 by "mach5one" with MS VC++ 15.5 (VS2017)

END.
```

LucianoZu, 2019-05-16:
I've read your log is not clear if your problem is on Windows 10 or 8.1. I'll try a connection on a Windows system asap, however I don't think it's an Ardulink bug. It's probably a problem on some driver on your system. Have you tried to use Arduino IDE to connect your PC with an Arduino board?

VTHMgNPipola, 2019-05-16:
Yes, I tried connnecting to my arduino board with Arduino IDE.
On the first try I was unable to connect, but after installing the correct drivers the Arduino IDE can connect to the arduino, but Ardulink still can't.

JayvanBruggen, 2020-05-21:
I'm getting about the same error on my Windows 10 system as  VTHMgNPipola  above. 
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=3368, tid=14832
I notice the pc===0x0000....   is the same number as his. 
Problematic frame:     [jSSC-2.8_x86_64.dll+0xb5db]

Could you please try to run an example like DataReceiver or ButtonQuest on a windows system because it is not working now and I was looking forward to it. 

When I try the example   DataReceiver I get
l\Programming\MeerJavaCode\ardulink\examples\DataReceiver>java -jar example-datareceiver-2.1.1.jar
WARNING:  RXTX Version mismatch
        Jar version = RXTX-2.2pre1
        native lib Version = RXTX-2.2pre2
Exception in thread "main" java.lang.RuntimeException: gnu.io.NoSuchPortException

It seems rxtx-2.2pre2.jar is packaged while another is expected.
So I've tried  ardulink-V2.1.1-20190517-0752.zip  for a while but I keep running into errors so for now I'll try an older version. If I can help let me know

pfichtner, 2020-05-22:
Just for reference: RXTX on master was replaced by https://github.com/NeuronRobotics/nrjavaserial (fork of rxtx). Maybe this fixes the problem regarding Windows 10. 
@LucianoZu is there any chance for users to access the CI builds as described in #28 to check out things in advance of the release

JayvanBruggen, 2020-05-22:
Possibly. I was trying three packed released versions from http://www.ardulink.org/download/ and they all failed in one way or the other. So today I will study some Maven and try to figure out how to get the whole build in my IntelliJ and take it from there. I suspect that can work, if it works for you.  I really want to and am going to make this thing work. Is it OK if I get back to you with troubles I run into?

LucianoZu, 2020-05-22:
Yes, of course

JayvanBruggen, 2020-05-23:
I have come to understand that the problem is with the JSSC library.  A google search for 
"pc=0x000000007110b5db"  without quotes , finds many projects and questions with this same error. This shows that the library is currently not working with Windows 10, so be aware that this is a dead end for now unless we find a fix for it. Currently working on finding a solution.