Ardulink/Ardulink-2 #52: OpenHAB Binding

PizzaProgram, 2020-03-31:
**Dear Developer,**
-
 Many enthusiastic people, like me are looking for a solution to connect more fix-wired IO ports + I2C sensors to the JAVA based [words #1 open source smarthome solution](https://ubidots.com/blog/open-source-home-automation/) called [OpenHAB](https://www.openhab.org/).
_(There are hundreds of topics where people tried and failed... )_

After 200 hours of digging forums, I have found your project, and as a programmer myself I have to say:
 - it's an amazing work!

What would you say, I would ask you in name of out community to help us integrate your project by [creating a Binding for OH](https://www.openhab.org/v2.3/docs/developer/development/bindings.html)?

There is a huge need to be able to connect cheap things like:
 - **Arduino NANO** boards or
 - Digispark AtTiny85 based dongles
 - many I2C GPIO extensions, like: **PCF8574** (8bit) and **MCP23017** (16bit GPIO) 
 - and other I2C sensors like temp, light, voltage, dust, proximity, etc
to make a smarthouse "work".

Width your skills I think it would be a peace of case ;)

Since I don't know anything about JAVA, I could only help with:
- testing
- promoting
- donating

What do you think? 
-
Do you have the mood + time to drop quickly something like that together?
_IMHO it would boost the popularity of this project a lot. ;) ..._

pfichtner, 2020-04-04:
What's your idea how to connect the microcontroller physically to some hardware? Would you like to connect the microcontroller directly to the computer that runs openhab? 
Several years ago i managed to control an microcontroller via openhab using MQTT. At that point in time, I had to use another machine acting as bridge between MQTT and the microcontroller since ardulink did not support the MQTT protocol by itself. 
@LucianoZu started to implement firmware for NodeMCU (ESP8266) some months ago. With this firmware you could control WiFi capable microcontrollers like ESP8266/ESP32 using openhab's MQTT binding. 
Could you please describe your requirement a little more in detail where a native Ardulink/OpenHAB binding would surpass the generic MQTT binding?

PizzaProgram, 2020-04-04:
Hi,
 Thank you very much for the answer!
Actually I have found an other library that is already doing everything I'm talking about:
[Firmata ](https://github.com/firmata/arduino)
It has a Node-Red connection too, which one has an OpenHAB sketch.

I won't say I wouldn't be happy to see +1 possibilities on the web instead of only one, so I'm still saying:
 - the IoT industrie is JUST starting to boom.

Currently OpenHAB does NOT have any "plug&play" possibility to connect all of these:
 - wifi
 - ethernet
 - USB
 - Modbus (RS485)
connected IO ports and sensor.

PizzaProgram, 2020-12-11:
After 8 month I'm back here. ;-)
As it turned out: Firmata is not really "[ready to use](https://discourse.nodered.org/t/arduino-i2c-firmata/31348/15?u=pizzaprogram)" for industry application nor for a simple house control.

I've learned following things I'd like to share:
- IoT is moving towards MQTT
- 75% of people using any kind of house-control are using NodeRed
- Even OpenHAB users are moving to NodeRed
 (Because aaaallllll the forums are full with ESP32 and other wireless things only.)

# So there is still a big need for a universal platform like this to "connect wired things together" !

Currently I'm looking a way to:
- upload a Firmware (.INO file communicating with Ardulink) to an Arduino Nano
- find out how to install Arduling to a (debian based) Raspberri Pi 
- try to communicate with I/O + analog + I2C ports of the board through Ardulink <==> MQTT

If you can provide any help with these, it would speed up these things a lot!

Thank you in forward!

_PS: Donated $10 a week ago. There should be a $1 $5$ 10 $15 $20 $25 $50 in the list instead of 1..10
I have probably less salary than you guys, but that's the only way I can help to move things forward..._

pfichtner, 2020-12-11:
> I've learned following things I'd like to share:
>     * IoT is moving towards MQTT
That's the reason why we added support for it :-)

>     * 75% of people using any kind of house-control are using NodeRed
So I am part of 75% of people :-)

>     * Even OpenHAB users are moving to NodeRed
>       (Because aaaallllll the forums are full with ESP32 and other wireless things only.)

Ugh!? OpenHAB and NodeRed are software, ESP32 is hardware, cannot follow your thoughts :-/

> Currently I'm looking a way to:
>     * upload a Firmware (.INO file communicating with Ardulink) to an Arduino Nano
The sketches resists in the sketch folder. You have to compile and upload them to your microcontroller (MC). Using Arduino IDE this is an easy task. 

>     * find out how to install Arduling to a (debian based) Raspberri Pi
Install JAVA, download Ardulink's lastest release. There are multiple programs that you can use then (UI, MqttAdapter, ...)
Please let us know how we could support you further! 

>     * try to communicate with I/O + analog + I2C ports of the board through Ardulink <==> MQTT
Ardulink only supports reading and writing from/to analog and digital ports of your MC. You cannot send I2C commands to hardware connected to the MC (firmata can do this). 

> _PS: Donated $10 a week ago.
Wohoooooo! Thanks for that! :-)