Ardulink/Ardulink-2 #54: Getting it to run in Windows 10, IntelliJ

JayvanBruggen, 2020-05-22:
Hi devs, I start a new thread here so other people may benefit from my/our endeavor. 
This is a follow up from our earlier discussions started by 	VTHMgNPipola. 
Running the pre compiled jars was unsuccesful so I decided to import sources and work from there, since I want to be programming with it anyway.

I have succesfuly imported ardulink-core-base   and ardulink-core-console  per Maven into my IDE.
I noticed Console is runnable with a main, and I am able to compile and run it. When doing this a start of a Jframe pops up and then I get this:  (I only post the first and last three erorr lines:)

java.lang.IllegalStateException: asciiString must not be null

at org.ardulink.util.Preconditions.checkNotNull(Preconditions.java:35)
at org.ardulink.util.URIs.newURI(URIs.java:18)
at org.ardulink.gui.connectionpanel.ConnectionPanel .getListCellRendererComponent(ConnectionPanel.java:91)

........................................lots more of these.........................

at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

What Java JDK are you using? I'm on 11, but I could set it to a lower one.
Else, what could be going on? I tried to trace what is throwing this or what the preConditions are, but the rabithole just runs to deep. I gave up after four calls deep.

JayvanBruggen, 2020-05-23:
It appears that the console is trying to display, and it is running. But on every mouse hover over the console, many many exceptions keep on throwing. After further inspection, it seems this has more to do with the Console graphical events and listening technique than with Arduino messages, so I'll skip Console for now and see if I can get something else to work

JayvanBruggen, 2020-05-23:
The problem with getting it to work on Windows 10 is with jssc.  The problem was worked on by some other people and this thread is interesting:  https://github.com/arduino/Arduino/issues/6788 
At the bottom there's a link to another jssc fork. Will try that now.
Also, can I use something else than Jssc? Can you suggest any alternatives? I want to use Arduino on Windows 10 through USB with Ardulink.

pfichtner, 2020-05-23:
The error you got happens when you don't have any available links (e.g. serial-rxtx, serial-jssc, ...) in classpath and was fixed few hours before you post this issue. So please update your sources using ```git pull```

JayvanBruggen, 2020-05-23:
Which error would that fix? Which code specifically should I update? I used maven to get Ardulink-core-base. I am not sure that I can use git pull from my IDE. Will it work if I reload the complete POM or is that not refreshed yet?  Could you please answer some more questions in my above posts? There are many. Please discuss more. Again: I am not trying to run Console anymore. But there is a big general problem with jssc on Windows10 in many projects of other people. So can I use something else? There is a new Jssc here : https://github.com/java-native/jssc/releases  . I tried using that but my guess is that Linkfacory or something near it does not register it like your own "ardulink-core-serial-jssc" that had jssc included (the crashing one)    Should I try working with nrrxtx?

JayvanBruggen, 2020-05-23:
I got it to work. I will post here what I did to give back to the community. After that I would like to make some suggestions to the developers of Ardulink, who made such an incredible and elaborate piece of software. 
If you try to follow my guide here, there is some liberty in your interpretation and you still need some experience, your route may differ from mine, but I hope to help. 
I started out not knowing what rxtx or jssc is. I just wanted to read analog pins from my Arduino to Java, and so found Ardulink. 
From the Ardulink website I downloaded and tried several of the release zip files, 2.1.1  2.0.1  and 2.0  and 0.6.1  but none of the  examples worked from the commandline for me. They all crashed. Mostly because a used library jssc-2.8.0  does not work anymore on Windows10.  --> AccesViolation. 
I did learn from the zip notes that a small sketch needs to be uploaded to the Arduino. I am using a Mega 2560. The text in the zip was confusing, it states that ArdulinkProtocol4LeonardoAndMicro.ino  is the most general sketch, but it is not. You should use  ArdulinkProtocol.ino , at least in the case of the Mega. 
Since the zip libraries didn’t work I decided to work with the sources and made an empty project in IntelliJ.  In ProjectStructure , project libraries I added some ardulink libraries using :   +     …  From Maven….  then type the word ardulink and in 15 seconds you can choose some ardulink libs from a list. Choose ardulink-core-base 2.1.1   and ardulink-console-2.1.1.  and   ardulink-core-serial-jssc-2.1.1  and import all the four things.  Download them to your empty project structure. 
In ardulink-console   there is a class called console, which contains a runnable ‘main’. In IntelliJ under run-edit configurations you have to set up an application : + .. application .. name it ‘YourConsole’ and for main class pick  or set   org.ardulink.gui.Console  You know, the class with main.  When I ran it it gave me a ton of error messages but soon after this was fixed by devs. 
Now the tricky part : Jssc is part of the library that does the usb com serial communication and was created by another dev and doesn’t work anymore for W10. But is is forked here : https://github.com/java-native/jssc/  and I downloaded  jssc-2.9.1.jar
Now in the downloaded Maven libraries I manually removed  jssc-2.8.0.jar  and files related to it and replaced them with the new 2.9.1  .  But leave the ardulink-core-serial-jssc-2.1.1.jar  alone.   To my big surprise, after this, even without registering in the application, ‘Console’ now worked and the exception_access_violation is gone.
Suggestions to the developers, to make the works more accesible to new users:
1  jssc appears not to work very well anymore, just google  pc=0x000000007110b5db    
please consider the new jssc 2.9.1 in the link above, that does work, and perhaps make it official. 
2   when people land on your website, there is no real clear article like Start Here, this is how it works. New people don’t know what rxtx and jssc is or know about the Link class. Things get really complicated if it doesn’t work. 
Only inside the distributed zips is first mention of the sketch to upload to Arduino. But it is not very clear whichsketch, or what the diferent sketches are for. 
The Arduino sketches are only in the distributed zips, they are not visible or mentioned on gitHub. They are hardly mentioned on your website. But they are essential. 
3 On Github there are many different folders called ardulink-something,  but it is very unclear which folders do what , or which one are needed for new user. But you need to be on gitHub since the release zips don’t work and have wrong versions inside and lots of confusing things. 
4 If we want to communicate with Arduino from java, what way  is best? There is rxtx which is not so good as far as I read, jssc which crashes, you mentioned nrrxtx or nrJavaserial but that doesn’t download from Maven/is empty. And then there is all these other methods like mqtt which seems to big if you just want some simple pin communication. No idea which one to use. I am just happy that I hacked myself into a working jssc 2.9.1. 
So perhaps can you give newcomers a more clear point of entry and some explanation of the different options. I love your work, but the presentation on the website is rather confusing and seems like a collection of articles, and you don’t know which one to read first if you are new.  There’s Ardulink 1 and 2, but nowhere what is the difference. The ‘Blink’ article doesn’t say how to set it up and is not good for starting. The Unique ID article is first but all we want is to get started.   I think the Ardulink messages article is very important so it shoul be near the top. Well, you get the point. So thank you for your work.

pfichtner, 2020-05-23:
@JayvanBruggen Many thanks for your reply, your compliments, comments, suggestions and researches! 

1. the fork of jssc with version 2.9.1 does not support mac os. Ardulink serial should work at least on linux (x86 and arm) and Windows. So just because the jar you mentioned works on your machine we cannot replace it without testing it on other platforms

2. Normally users should not care about how ardulink is structured internally. For the sketches you are absolutely right, @LucianoZu we should provide a HOWTO describing a.) how to prepare the Arduino and b.) which jar contains a main (runnable jar) and what it is for

3. Again, Normally users should not care about how ardulink is structured internally but beside that we could add a README.md for each submodule. I cannot follow your argumentation. We serve our sources on github. There is no need for us to be on github. And I don't see where the release zips are not working. Could you explain? 

4. When you want to communicate using a PC the normal case is that you do this via serial. So you need a java library that is capable to do. At the moment Ardulink supports two different libraries at the moment rxtx and jssc. rxtx already is replaced by nrrxtx in master and so will be available with the next release. proxy/mqtt/rest are modules if you want to to communicate with the PC where the arduino is attached using another PC. 

So you're right, we should have some articles describing the main areas of operations to support newbies and because they are missing. 
Ardulink-1 is the old (first) implementation of Ardulink's ecosystem and not longer maintained. @LucianoZu perhaps we could purge it? 

Good starting points are the example projects, e.g. the example-datareceiver. When checked out you should have everthing in the IDE to play around with Ardulink. 

Thanks for your reply!

JayvanBruggen, 2020-05-24:
Hi Peter,
  thanks for your kind reply. 

1  I understand about the jssc OS support decision. However, seeing it from the windows side, there are a lot of comments on how unstable rxtx is. So when we see that jssc is better we want to use that.
Besides, I tried rxtx. But one naturally starts with the zip releases. I tried the three latest 2.x.x. versions. I ran them from the command line in the correct way. Anything using jssc crashes with the mentioned access violation. And anything rxtx crashes with a rxtx version mismatch included in the zip. So you see, the releases don't work for Windows 10.
As a newcomer, at that moment you have no idea what rxtx is, and you are not able to replace it because you are just running a release demo jar to check out the solution. This can be intimidating enough to look for another solution. But I decided to dig in really hard.

2&3  We agree, I don't really need to know much how Ardulink is structured inside. That would be the beauty of using a library. I want to know just enough to get readings and drive commands working. Perhaps there is a misunderstanding, you wonder why I look at the gitHub structure? Well I didn't really want to, but with the zips failing you have to work with the sources. This directed me to gitHub quite naturally : On the Ardulink website there is a download menu with two options: the releases, and sources, which  lands you directly on gitHub. Also, on the website people are asked to comment and ask questions on gitHub. So new (Windows) users are directed there you see.  This line at the bottom of gitHub was essential: "Recommended way of using Ardulink-2 ... is declaring a dependency on “ardulink-core-base” ...."

4 I am glad to hear that the use of nrrtxt will be official. I agree that my use of jssc 2.9.1 is a little shaky and I would rather know that something dependable is supported. So in the future I will make use of nrrxtx. When I tried it this time there was a problem, which is not relevant now. Glad to hear it will work in the future, when it is bundled correctly. I will test for you if you ask me, on an empty VM.
Just out of interest:  so with proxy/mqtt/rest  , you would connect Arduino on one end with  say nrrtxt to a pc, with also mqtt on it. And then in another house using mqtt I would control it, right? Or how would you attach mqtt to Arduino? Still using serial right?

Yesterday I got to read some AnalogPin readings on my computerscreen in Console,  so I made enough progress. Coming weeks I'll do some programming, I may come back with some questions, and I will show you the practical results of the battery project.

Thanks!

JayvanBruggen, 2020-06-14:
Hi, today I tried to follow the tutorial   http://www.ardulink.org/how-to-manage-input-from-arduino/
which seemed like a good starter. It turns out this tutorial program tries to import a lot of classes that are supposedly from Ardulink 1.  For example SerialConnectionPanel and DigitalPinStatus.  After closer inspection I decided this example program is too far away from the current library 2 to be feasible. It seems not that useful now. It would be great to have a little demo program like that there for Ardulink 2.

LucianoZu, 2020-06-14:
Yes,
It's for Ardulink 1. The site has two TAGs: Ardulink 1 and Ardulink 2.

Luciano

JayvanBruggen, 2020-06-14:
Hi, I now have   example-datareceiver working and am nearly ready to continue to develop my own battery application.  But I hesitate a little to start to build it uppon the Ardulink technology. Because even the datareceiver demo, is now using the jssc2.9.1 that we talked about. This is the only way I found to get it to work on windows 10. But you do not support this 2.9.1.   I would appreciate it if you could let me know a little more about the upcoming   nrrxtx   implementation. So that when unsupported  jssc 2.9.1  stops working I can fall back on the supported and newer  nrrxtx.  Otherwise I may get stuck with my software and a waste a lot of work. 

Also, I write to you here because the Ardulink website contact button directs me here. If I could PM you in anyway I probably would. 

Thank you

pfichtner, 2020-06-16:
Ardulink is designed completely modular. Even if Ardulink would not support **any** serial connection java framework you could add whatever you like within minutes. So there **never** is the danger of wasting time and to get stuck! 

With the next release of ardulink we will replace rxtxt with com.neuronrobotics:nrjavaserial:5.1.1 (done on may 05/10/2020 with commit 888c0f3a6bda663509a638440c3473a4df40edf9) so already available in the code base. 

We do not support jssc in newer versions because of several reasons: 
- jssc does not support mac os anymore
- version not available an maven central (yet?) (not a showstopper since we could include the jar in a local repo but I'd like to avoid that)
- we do not have made a release within the last weeks

Since the the contact form http://www.ardulink.org/contact-us/ seems to be broken (@LucianoZu what's wrong there?) and I learned that "error creating database connection" is in dutch " Fout bij het maken van de databaseconnectie" on www.jayvanbruggen.nl , please let me know how I could contact you, so I'd be able to support you using mail or even possibly a by making a video call.

JayvanBruggen, 2020-06-17:
Your reply looks promising. I wouldn't know where or how any communication system like rxtx or jssc exactly makes contact with / interfaces with Ardulink. But since you and Luciano are still quite active on the project I will continue with Ardulink and not worry about that aspect. 
Incidently I recently cancelled my website jay....nl registration, so that's why that webadress doesn't work. You can contact me on   jay at audionstudio . nl  at = @ . In a couple of days I could use some directions, thanks for offering.

pfichtner, 2020-06-29:
@JayvanBruggen did you receive any of my mails?

pfichtner, 2023-08-06:
Had a video call with jay in 2020, so should be done. Please feel free to reopen if not.

closed 2023-08-06 by pfichtner