Ardulink/Ardulink-2 #56: Doc / WebPage improvements

PizzaProgram, 2020-12-12:
**Hi,**
 I'm writing down my steps and thought to show, how a "new user" (like me) approaches things on your webpage to get things going.
( _Please consider: I have 35 years of iT+programming experience, 2y home-control specialisation, etc... so if I'm not finding something easily, than probably others with less experience won't either..._  )

1. I'd recommend to start ardulink.org with the current "What is..." page
2. "Home" should be rather renamed to "News"
3. "What is" (the new homepage) should be improved with:
- a few pictures 
 -- from some Arduino boards + 
 -- from the Arduino IDE + 
 -- from your java app
- a "Getting Started" infos: 
 -- Here you can download the .INO files  ... 
 -- instractions to install it to Windows / linux ... watch video
 -- first "blinking-Led" example
 -- MQTT example ...

4. The "Compatible Boards" section should be rather  a long list, than 2 simple lines:
 - Digispark  (USB)
 - Arduino UNO (USB / serial)
 - Arduino micro 
...
 - ... and  theoretically ALL other Arduino or compatible boards ! 
This will help with google search too.

PizzaProgram, 2020-12-12:
It was not easy to find those .INO files! They are very well hidden in sub...sub...sub folders.
Without your last hint in my prev. "issue" topic here #52 it would have been nearly impossible. 
Lucky me (after 2 years of github using) just realised there is a "Goto file" button I can use to filter for "sketch" or ".ino" ...

But that part needs serious improvement.
New users will definetly need a better "step by step" instruction-page with pictures about where to download.

Maybe these files could be part of the IDE examples? Or at least as a downloadable library?
That would boost popularity a lot!

PizzaProgram, 2020-12-12:
To change unique ID of the board should be included in the .INO files on the top, not as a separated blog !

# Identifying which board is which one is essential for safety !

So this part needs improvements too. 
Also I'd like to recommend to implement on your "master side" a safety-protocol to check if you are connecting to the board name you should ?

 ( _Think about for a moment what happens, if 2 USB ports getting exchanged, and the instead of a light turns on >> a metallic roller-shutter goes down and cuts off a child's head? Or other industrial applications ... ?_ )

PizzaProgram, 2020-12-12:
It is not clear:
- is it possible to refer to a slave board from the master side by unique-name?

(And not as a fixed IP addr. or USB port?)
Maybe with MQTT?

pfichtner, 2023-09-10:
I would like to recreate the entire documentation based on github pages, which would also include a reorganization of the firmware stuff, e.g. an installer like https://install.wled.me/ would make the process much easier for users.

LucianoZu, 2023-09-10:
I agree, web site too should be renewed or substituted by another documentation tool. Maybe site will go down soon.

pfichtner, 2023-09-23:
Technical initial setup done: 

- https://github.com/Ardulink/Ardulink.github.io (source)
- https://ardulink.github.io/ (will be automatically build on changes in the Ardulink.github.io repo