Ardulink/Ardulink-2 #79: [security enh.] Setting board name / password

PizzaProgram, 2022-08-24:
To prevent:
- multiple boards getting mixed up or
- "intruder board" getting plugged to the system

it would be essential the add basic name + password for safety.

.INO example:
`name="" // fill this to force-check and enable connection. If empty = not safe = not checked.`
`psw="" // Security check. Fill this with SHA256 hexa format of the password you are sending. If empty, check will be skipped.` 

[ x ] Also force Firmata group to implement it the same way.

PizzaProgram, 2022-08-24:
Example for password generation:
psw="DC1E7C03E162397B355B6F1C895DFDF3790D98C10B920C55E91272B8EECADA2A";
// SHA256 Hash of "MyPassword" See: https://coding.tools/sha256

pfichtner, 2023-04-30:
Could be combined with the UUID feature. 
- UUID is written to EEPROM (already existing sketch) and is used as secret
- The sketch has to be extended to support some "unlock" command where the UUID has to been send by the Ardulink (PC). Until unlocked the sketch should refuse every interaction with the Arduino
- Add support to Ardulink to send the "unlock" command

PizzaProgram, 2023-05-01:
Do you mean the UUID would be the "password" part? 
 - If yes, good idea! :+1: 

IMHO a unique name should be better to identify the board. Like: "basement2", "spaceship-door3", "[breath-mashine](https://trikoot.net/bcmachine/)", ...

It would be preferable to have an easy-editable, unique field for that!
 _(Because it's much easier to change 1 number or word before flashing each board. `name="basement4 ... 5 ... 6"` )_

**PS:** And maybe later we could develop an "auto search of available boards" function, so these names should be public.

pfichtner, 2024-12-01:
@PizzaProgram I added a password POC, see [this branch ](https://github.com/Ardulink/Ardulink-2/tree/board-password)
You could either check out the branch or if you want to try the binaries you can download the jars from [the CI run](https://github.com/Ardulink/Ardulink-2/actions/runs/12105883656) (only available for a limited time)

You have to set the secret in the sketch (ArdulinkProtocol) and set it when connecting via serial using the "secret" attribute. Of course you can use any String you like (e.g. as hash)

Any feedback welcome

pfichtner, 2025-01-06:
@PizzaProgram Is there still interest in the feature? is it possible to try it out?

PizzaProgram, 2025-01-08:
# [OFF]

Dear @pfichtner ,
I've just checked the whole code change, line by line [here ...](https://github.com/Ardulink/Ardulink-2/compare/master...board-password)
I have to say, You did an excellent job! Never have thought it would be such a big work, and so many places had to be rewritten.

Sadly I can not test it myself currently, since I did not start working with Ardulink two years ago.
I was looking for a way to extend a Raspberry Pi to be able to handle more pins a few meters away via USB. _(Maybe add some temp readers later 10-20 meter away.)_
The main problem is still, that almost each "link-project" available on the internet is calculating with +1 Arduino / I2C / etc ... board, and not with multiple ones.

## So you can not distinguish these boards safely from each other!

Whenever there is a power outage, or restart, the USB/i2c numbering getting mixed up.  
What can be very dangerous. Even at a simple home-control system. 
_(Like: overheating water instead of turning on a lamp.)_ 
Not to mention industrial environment...

That's why I have recommended, as primal directive to be able to :

## Give NAMES to boards

... and while doing this, maybe a password too, as secondary objective, because there are always people working with these things, who can:
- make mistakes,
- or even trying to harm.

Currently I'm working with "[RPi Pico](https://www.raspberrypi.com/products/raspberry-pi-pico/) boards" + "[Configurable Firmata](https://github.com/firmata/ConfigurableFirmata) firmware" + "[Nore-RED Arduino module](https://nodered.org/docs/faq/interacting-with-arduino)".
It is not ideal, because, the NR module is very very basic, full of bugs and non-implemented things, but this method has one stable side:
 - You can connect to the boards thought it's unique built-in serial name!

[image]

Something similar would be the ideal goal for Ardulink too, when all boards would be listed by name and easily choosable.

PS: Sorry for the late response.

pfichtner, 2025-01-08:
@PizzaProgram thanks for your detailed answer, I will read and answer it later at my leisure

pfichtner, 2025-01-08:
@PizzaProgram just a short aspect on your last point: 
`/dev/serial-by-id` is not a feature of node-red but of linux, so confusing COM1 with COM2 was never an issue when using linux (but windows). 

Ardulink does not show `/dev/serial-by-id`  because these ports do not match the rexep
`PORTNAMES_REGEXP = Pattern.compile("(ttyS|ttyUSB|ttyACM|ttyAMA|rfcomm|ttyO)[0-9]{1,3}");`

I guess it would be not much more to change this regexp to show `/dev/serial-by-id` links in the combox box for selection in Ardulink as well. So if this would help you: This would be a very, very, very low hanging fruit.

pfichtner, 2025-01-08:
One line of change

[image]

pfichtner, 2025-01-14:
@PizzaProgram would this "enhancement" help you in any way?

PizzaProgram, 2025-01-16:
Dear @pfichtner ,
 Thank you very much for all the attention of this topic and the changes you are doing!

1. Yes, it would help a bit, if ` /dev/serial-by-id` would be choosable.

2. Sadly not for windows. _(Personally I'm using Raspberry with Debian based RPi OS, but others may not.)_

3. As I wrote earlier: safety is very important, especially in industrial environment, so I still think adding *names* to the boards, prevents them to mix them. 

4. I know, that writing a subroutine, that would be difficult:
    - search all ports automatically,   
    - identifies all boards running Ardulink
    - gets the names of the boards and offers them to users 
    - and if a board is getting disconnected during operation, it would periodically re-search for it, maybe it was re-plugged in an other port.

6. I'm not asking for these changes for just myself, but for all of mankind, because there is really not much possibility out there to control small boards safely.

### Example: _Think about following scenario:_   
- An industry-PC gets a power surge and 2 of 4 USB ports fried. 
- There are 2 Ardulink boards plugged in, but the worker is realizing 1 is still working = blinking.
- He quickly re-plugs the other board next to the working board's USB pair, 
- and in a few seconds everything works again.
- The support team gets a detailed report from the Logs, telling the board named: `main-robot-arm`  was unconnected + reconnected, not just something like: ` "...usb1-a86..."` which is very hard to identify. _(Especially, if the man originally installed the board is not there any more to remember. Which is an often case in companies.)_

_(Something similar happened to me, 2 days ago, in one of the pizzeria I support, with thermal printers.)_

Now this example is only about 2 boards. What if there are USB HUBs, and 8+ boards involved?

pfichtner, 2025-01-17:
As a "quick fix" which is not related to Ardulink but that would easily fix your concern you could create a custom udev rule

e.g. `/etc/udev/rules.d/99-usb-devices.rules`

`SUBSYSTEM=="tty", ATTRS{idVendor}=="XXXX", ATTRS{idProduct}=="YYYY", SYMLINK+="main-robot-arm"`

So you could access main-robot-arm as `/dev/main-robot-arm`
Of course this only works as long as you can distinguish in vendor and product level. If all devices have the same vendor and product id this approach won't work.

pfichtner, 2025-01-24:
With ce5fb7cc48448a3e2b3d24a37493646b512614fb I just did add a preview feature that will add `/dev/serial/by-id/` links as choosable/valid serial links. 
This feature has to been enabled by setting `with.serial.links.byid` environment variable to a not empty value, e.g. 
by using `-Dwith.serial.links.byid=y`
**This only works when using jssc at the moment!**

PizzaProgram, 2025-01-24:
What is JSSC ?

> One line of change
> 
> [image]

Which program is this picture from?

PizzaProgram, 2025-01-24:
> As a "quick fix" which is not related to Ardulink but that would easily fix your concern you could create a custom udev rule
>
> e.g. `/etc/udev/rules.d/99-usb-devices.rules`
>
> `SUBSYSTEM=="tty", [...]

Thanks for sharing this! Very cool, you can do these!
_(Sadly average users, like me and other 99.9% can not.)_

pfichtner, 2025-01-24:
> Found JSSC. https://github.com/scream3r/java-simple-serial-connector
> 
> How to apply to Nore-Red? (Never worked with Java before, only JS)

You can't "apply" jssc or Ardulink to Node-Red. 
What exactly would you like to achive with Node-Red? 

I have a Node-Red PoC where I control an Arduino connected to a PC (e.g. raspberry pi) using MQTT. 
So I start Ardulink-MQTT on that PC and control the Arduino by sending MQTT messages to the topic `home/devices/ardulink/D2` e.g.

> > One line of change
> > [image]
> 
> Which program is this picture from?

That's Ardulink's Console

pfichtner, 2025-01-24:
> The main problem with this solution:
> 
>     * it will not work, if you plug the same board to a new PC.

That's true but the question is how often this will be needed. 

> That's why I've suggested to hard-code names to the boards, that Ardulink would read, and offer to the users to choose from a list of boards.

I don't want to optimize Ardulink for the rare case that the user has little work to do when replacing the PC. The effort to extend the protocol and the firmware and to scan all serial ports for Ardulink boards is not in good proportion to creating the udev rule again for a new computer. 
If someone finds the time for implementing it: I am happy about a PR!

pfichtner, 2025-01-25:
I did add a node-red example to the [samples](https://github.com/Ardulink/Ardulink-2-Samples/)

https://github.com/Ardulink/Ardulink-2-Samples/tree/main/node-red